ucb-bar/gemmini #229: Support for non-zero padding

fPecc, 2022-07-22:
Hi all!

I was interested in knowing if you are planning on supporting non-zero padding for the LoopConv state machine. 

**Use case**

TFLite models quantize the input of a layer with a scale and an offset. Because of this, in order for the layer's convolution to make sense mathematically, the padding zeros also need to be quantized with the same scale and offset. Because the offset is non-zero, the resulting padding value is also non-zero.

**Current solution** 

Pad outside Gemmini, and then call tiled_conv_auto with padding = 0, with the new padded IN_DIM (this is the original IN_DIM plus the padding). This works fine mathematically, but is clearly slow.

**Proposal**

To add the padding value as an SInt(8.W) to the gemmini loop ws CISC instruction, and then modify the LoopConv to inser this padding accordingly.

hngenc, 2022-07-27:
No plans to implement this in the next few months. But after the ASPLOS deadline (in October), I might have more bandwidth to add more features.

Sorry about that! I'll keep this issue open till we implement this.

fPecc, 2022-09-13:
Hi @hngenc ,

I ended up implementing it because I needed it for a use case, I will prepare the pull request so that we can review it together.

hngenc, 2022-09-13:
That's wonderful; looking forward to it!

fPecc, 2022-12-01:
Hi @hngenc , sorry for the delay, had a couple of deadlines I needed to achieve. I am now fully migrating what I did for the new Gemmini version, once I have everything working with the new version, I will let you know

hngenc, 2022-12-02:
Sounds great; thank you!