ucb-bar/gemmini #230: Gemmini implementation: misunderstanding of Serial_tl port usage

deykid, 2022-07-27:
Hi everyone,

I am looking to implement Gemmini on an FPGA (zcu102). I have generated a project with a rocket core to host Gemmini, but I have some questions.
I wonder about the use of serial_tl ports in my case. Indeed, I looked at the signals through GTK Wave and I saw that serial_tl ports are used. However, I am not able to find exactly in the test bench where the sent data are defined. Also, I found some explanations about tilelink or SerDes in the chipyard documentation, but I'm not sure what these ports are used for in my case.

- https://github.com/ucb-bar/chipyard/blob/main/docs/Advanced-Concepts/Chip-Communication.rst
- https://github.com/ucb-bar/chipyard/blob/main/docs/Generators/TestChipIP.rst

Could someone give me more explanation about how these ports should be use, please?

[image]

deykid, 2022-08-19:
Hi,

Does anyone know where the sequences for serial_tl and jtag are defined and executed?

Best regards,

jerryz123, 2022-08-19:
chipyard/generators/testchipip contains the TestHarness-side serial adapter interface, which plugs into serial_tl.
chipyard/generators/rocketchip contains the definition of SimJTAG, which plugs into the JTAG interface.

closed 2022-08-19 by jerryz123