ucb-bar/gemmini #231: Not able to generate "simulator-chipyard-CustomGemminiSoCConfig-debug".

joychauhan-sudo, 2022-07-29:
While running "./scripts/build-verilator.sh –debug"  command to generate waveforms, the "simulator-chipyard-CustomGemminiSoCConfig-debug" is not generating.
I have generated this folder by running below command manually:
make ${debug} CONFIG=CustomGemminiSoCConfig
Is there any other way to solve this issue?

fakoberer, 2022-07-31:
./scripts/build-verilator.sh --debug
After executing this command you should see a lot of stuff in the CHIPYARD/sims/verilator/generated-src/CustomGemminiSoCConfig   Do you see them? If you don't that means you were not able to create the simulator environment for Gemmini config in verilator. I don't know the details of the build-verilator.sh script. However, you seem to be using this line incorrectly. See the difference: "--", "–". Execute that line again and that should take around 30 min-45 min. Then you can run the test softwares on verilator.

joychauhan-sudo, 2022-08-01:
Yes, I am able to see lot of stuff in this path "CHIPYARD/sims/verilator/generated-src/CustomGemminiSoCConfig" by running "./scripts/build-verilator.sh --debug" command (avoid the typing error "--" & "–"). But to observe waveforms I need "simulator-chipyard-CustomGemminiSoCConfig-debug" file which is present in run-verilator.sh and this file should generated by build-verilator.sh script but not generating it, can you confirm this by running "./scripts/build-verilator.sh --debug" command ?
I have generated this file manually by running below command:
make ${debug} CONFIG=CustomGemminiSoCConfig

fakoberer, 2022-08-01:
I (not 100% sure) think the command "make ${debug} CONFIG=CustomGemminiSoCConfig" you wrote in the sims/verilator directory does not build a complete Gemmini chip and simulator because there are other parameters which are set for rocketchip as default. What you should is nothing but running the following command. If that does not help, we should wait for experts.
[image]

joychauhan-sudo, 2022-08-01:
Ok thanks for reply, I'll check on my system again.

closed 2022-08-01 by chauhanjoy