ucb-bar/gemmini #232: cannot build spike for gemmini

fakoberer, 2022-07-31:
Hi to all!
Before starting, I want to state that I am a newbie to this whole GitHub thing.

Although I installed Risc-V tools and esp tools, I am unable to install spike for Gemmini. 
I can successfully run the spike for bare-metal Risc-V codes just like in the following link: https://chipyard.readthedocs.io/en/stable/Software/Baremetal.html

However, I am unable to run the command since I have no "build" directory in the direction ./riscv-isa-sim/
[image]

And I could not run the following command as well:
[image]

P.S.  I successfully build a simulator environment on Verilator and I ran test softwares on it. There is no problem with the verilator.

Thanks in advance for your help.

deykid, 2022-08-01:
I'm not an expert, but maybe I can help find the error. On my side, everything works fine.

Have you installed all the dependencies? (SBT, Poky, Qemu, Firemarshal, DTC, Git >= 2.17, Verilator)

Except for the dependencies, I have only run this:

```
git clone https://github.com/ucb-bar/chipyard.git
cd chipyard
git checkout 117624d8eea27bafd613eec09e9b9b3e31239e08
./scripts/init-submodules-no-riscv-tools.sh
./scripts/build-toolchains.sh esp-tools
source env.sh
cd generators/gemmini
git fetch && git checkout v0.6.4
git submodule update
```

Did you run the two scripts for "init" and "build" the toolchains?

fakoberer, 2022-08-02:
I figured it out. Thanks a lot for your contribution. There was an error with esp tools which I was ignorgin ":)" preventing it installing the esp properly. Problem solved.

Teut0nic, 2022-08-03:
I'm having problems with this part too.  I'm getting errors when running the './scripts/build-toolchains.sh esp-tools' script.  The last error text that shows for me is "Makefile:213: recipe for target 'stamps/build-glibc-linux-rv64imafdc-lp64d' failed"  How did you get around your errors for this script? Thanks!

fakoberer, 2022-08-03:
> I'm having problems with this part too. I'm getting errors when running the './scripts/build-toolchains.sh esp-tools' script. The last error text that shows for me is "Makefile:213: recipe for [...]

I am sorry I have no knowledge regarding your problem.

Teut0nic, 2022-08-05:
> I figured it out. Thanks a lot for your contribution. There was an error with esp tools which I was ignorgin ":)" preventing it installing the esp properly. Problem solved.

What was the error you were getting?

fakoberer, 2022-08-05:
> > I figured it out. Thanks a lot for your contribution. There was an error with esp tools which I was ignorgin ":)" preventing it installing the esp properly. Problem solved.
> 
> What was the error you were getting?

I really do not remember clearly right now. But at the beginning of the error it said that some program could not . been found. So I basically run $sudo apt-get-install <that program>

hngenc, 2022-08-10:
@Teut0nic

If you're having issues with the `./scripts/build-toolchains.sh esp-tools` step, then it might be worth asking about it on the Chipyard mailing list, as that isn't a Gemmini-specific problem.

Are there any other error messages you found in the step? You might want to save the full output to a log file, and then search for error messages in there.