ucb-bar/gemmini #234: Running C++ Code on Gemmini

fakoberer, 2022-08-05:
Hi All!
I was wondering if I could run a C++ code on CustomSoCCGemminiConfig in either Verilator or Spike. So, I searched on the Google to see if I could compile a C++ program for Risc-V. I could not find anything useful though. I want to ask if it is possible to do so. If it is possible, what steps should be followed ?

hngenc, 2022-08-10:
I actually did this years ago, but I never merged it into the main branch. Are you able to run the C programs in `software/gemmini-rocc-tests` (as described [here](https://github.com/ucb-bar/gemmini#software)?) If so, you should be able to make small changes to `software/gemmini-rocc-tests/bareMetalC/Makefile` in order to compile C++ programs instead of C programs.

May I ask why you want to run C++? Are you hoping to take advantage of templates or some other C++-specific features?

fakoberer, 2022-08-10:
Hi Hasan!
Yes I am able to run those test C codes. I did minor changes on those tests and recompiled and ran them. There is no problem with C. I managed to compile a C++ code using riscv64-unknown-elf-g++ and I can run it on default (rocketchip) config. But my intend is to run a basic image processing code with Gemmini accelerator. Now I am searching for C++ image processing libraries. 
At this point, what do you suggest me ?

hngenc, 2022-08-10:
I'm not sure which image-processing library would be best for your needs — I'm mostly just familiar with OpenCV, but I imagine that cross-compiling it for RISC-V would be difficult :)

Also, just be aware that it can be very difficult to run code with lots of library dependencies on Verilator/Spike, because the baremetal RISC-V libraries that we use in `software/gemmini-rocc-tests` are quite limited. (They don't even fully implement `printf`). You might find it easier to run such programs on Firesim, because then you don't need to use the baremetal infrastructure, but can just cross-compile RISC-V programs as you would normally do.

fakoberer, 2022-08-10:
I see. To be honest, I was expecting such a comment. So, are you recommending running linux on a custom config and running the codes in linux? And also, would not just including the image processing library and compiling it via riscv64unknwonelf work?   

In other words, I am a bit confused here. You guys have created an accelerator, but it is very hard to use it fully functional. What is the best way to take advantage of your hardware? 

Thank you so much for your quick reply.

hngenc, 2022-10-11:
When you simulate Gemmini on Firesim, you’ll be running on a
fully-functional Linux kernel, with a fully-functional software stack. That
makes 3rd-party software compilation a lot easier.

Verilator/Spike are really just meant for testing and profiling, rather
than complicated “real-world” program.

Regardless though, if your image-processing library is just pure C++, and
uses no 3rd-party dependencies (and ideally even avoids using the “math.h”
library), then it might compile perfectly fine on the baremetal software
stack, which you can run easily on Verilator. It’s definitely worth trying.