ucb-bar/gemmini #239: Error running build-verilator.sh and run_verilator.sh

zhongwujie, 2022-08-25:
I have followed the instructions and built the toolchain successfully.
```
git clone https://github.com/ucb-bar/chipyard.git
cd chipyard
git checkout 117624d8eea27bafd613eec09e9b9b3e31239e08
./scripts/init-submodules-no-riscv-tools.sh
./scripts/build-toolchains.sh esp-tools

source env.sh

cd generators/gemmini
git fetch && git checkout v0.6.4
git submodule update

cd -
cd toolchains/esp-tools/riscv-isa-sim/build
git fetch && git checkout 090e82c473fd28b4eb2011ffcd771ead6076faab
make && make install
```
However, when I run build-verilator.sh, I get the following errors.
[image]

When I run run_verilator.sh, it shows that there is no such a config file.
[image]

zhongwujie, 2022-08-30:
I wonder if the ubuntu version matters. My ubuntu version is 20.04 while the chipyard document suggests 16.04 and 18.04.

hngenc, 2022-09-11:
Hmm, usually `Error 137` is caused by an out-of-memory error. Can you try re-running on a machine with more RAM?

zhongwujie, 2022-09-16:
Thanks. After I set the virtual machine RAM to 8G, it compiled successfully.

closed 2022-09-19 by hngenc