ucb-bar/gemmini #240: How to generate the waveform of some modules but not all of the system?

hust-gaoyujing, 2022-08-28:
When i used the command `./scripts/build-verilator.sh --debug`  to simulate and generate some waveforms, i found the waveform files (.vcd) is so big (maybe more than 100G) that it's so hard for my compute to browse waveforms. So i have a question, can gemmini only generate the waveform file(.vcd) of some modules (that i need) and how to do it? I think that's maybe significant and useful for us to debug.

Thanks!

hngenc, 2022-09-11:
If you have a VCS license, then you might want to use that instead. VCS generates `.vpd` files, which are _much_ smaller than `.vcd`.

For those who don't have a VCS license, I agree this would be a very useful feature. I'll look into adding this; but it might be a little while till I can get around to it myself.

hust-gaoyujing, 2022-10-19:
I think it may be addressed in  line 284 - 298 in `chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/emulator.cc` 
as follow:
```
  Verilated::randReset(2);
  Verilated::commandArgs(verilated_argc, verilated_argv);
  TEST_HARNESS *tile = new TEST_HARNESS;

#if VM_TRACE
  Verilated::traceEverOn(true); // Verilator must compute traced signals
#if CY_FST_TRACE
  std::unique_ptr<VerilatedFstC> tfp(new VerilatedFstC);
#else
  std::unique_ptr<VerilatedVcdFILE> vcdfd(new VerilatedVcdFILE(vcdfile));
  std::unique_ptr<VerilatedVcdC> tfp(new VerilatedVcdC(vcdfd.get()));
#endif // CY_FST_TRACE
  if (vcdfile_name) {
    tile->trace(tfp.get(), 99);  // Trace 99 levels of hierarchy
    tfp->open(vcdfile_name);
  }
#endif //
```
And i also find the solution to resolve this question in verilator user guide[https://www.veripool.org/ftp/verilator_doc.pdf],the method B may be appropriate.But i just do not know how to put it...

```
8.1.12 How do I speed up writing large waveform (trace) files?
A. Instead of calling VerilatedVcdC->open or $dumpvars at the beginning of time, delay calling it until
the time stamp where you want tracing to begin.
B. Add the /*verilator&32;tracing_off*/ metacomment to any very low level modules you never want
to trace (such as perhaps library cells).
C. Use the --trace-depth option to limit the depth of tracing, for example --trace-depth 1 to see only
the top level signals.
D. You can also consider using FST tracing instead of VCD. FST dumps are a fraction of the size of the equivalent
VCD. FST tracing can be slower than VCD tracing, but it might be the only option if the VCD file size is
prohibitively large.
E. Be sure you write your trace files to a local solid-state drive, instead of to a network drive. Network drives are
generally far slower.
```

 I will appreciate it if someone can solve it!

hngenc, 2022-11-23:
You might want to look into adding a `+permissive +dump-start=X +permissive-off` option to [this line](https://github.com/ucb-bar/gemmini/blob/master/scripts/run-verilator.sh#L91). That will delay generating the VCD file until a `X` number of cycles of passed. That should make simulations faster, and VCD files smaller.

Just a warning that I haven't tried this myself, so there may be a bug in that line.