ucb-bar/gemmini #241: Is there a document depicting `config_norm` in branch "mlsys-tutorial-2022"?

hust-gaoyujing, 2022-09-07:
I am so interested in the new thing "norm" when i read the codes in branch  "mlsys-tutorial-2022", but i can't find a document for it to understand its function and how to use it. So i want to know whether or not there is a document depicting `config_norm` ?

hngenc, 2022-09-19:
We're planning to merge that in Gemmini's `master` branch in the next few weeks, along with documentation. We are currently just distracted by some upcoming paper deadlines :)

`config_norm` provides support for LayerNorm operations and [I-BERT's integer-only softmax approximation](https://arxiv.org/pdf/2101.01321.pdf). These are both essentially normalization operations where the innermost dimension is the one being normalized. However, we hope soon to extend `config_norm` to also support operations like BatchNorm, where the normalized dimension is the outermost dimension.

hust-gaoyujing, 2022-09-19:
Thanks for your reply. Your works are fantastic and instructive to who are instrested in accelerator.Looking forward to more of your  works.

hngenc, 2022-09-19:
Haha, thank you! :)

ybai62868, 2022-09-19:
Great work and thanks again! Looking forward to seeing the new slides and records in MLSys-2022-tutorial !