ucb-bar/gemmini #242: gemmini_loop_conv_ws

minmin20028, 2022-09-16:
Hi.
I wonder what the **_gemmini_loop_conv_ws_** ISA`s role is.

In **_sp_tiled_conv_** function,  There is only **_gemmini_loop_conv_ws_** ISA.
And below that, the whole processes of mvin, compute, and mvout etc are commented out.

Do **_gemmini_loop_conv_ws_** and the whloe process do the same thing?
Or does using just one **_gemmini_loop_conv_ws_** makes whole process more efficient or fast?

Since there is no such instruction like **_gemmini_loop_conv_ws_** for depthwise convolution, 
I`m wondering if making such instruction for DW conv would make it work more efficiently.

Always thanks for your work.
Best,
Mingeon.

hngenc, 2022-09-19:
The commented-out code does the exact same thing as `gemmini_loop_conv_ws`. We just keep the commented-out code to help with debugging and as a form of documentation.

The `gemmini_loop_conv_ws` instruction dynamically schedules the `mvin`, `mvout`, and `compute` instructions to maximum the overlap of matmul and data-movement operations. The instruction also performs double-buffering, and some other optimizations to improve performance above what the commented-out code can achieve (even though both of them _functionally_ do the same thing).

You can look in `LoopConv.scala` or `LoopMatmul.scala` to learn more about how these "loop-unroller" instructions are implemented.

minmin20028, 2022-09-23:
Thanks for your reply.

closed 2023-01-28 by minmin20028