ucb-bar/gemmini #245: nBanks/nWays error when compiling Gemmini

hchenji, 2022-10-07:
Using chipyard 1.8.0 and gemmini 0.6.4, I get this error when I run build-vcs or build-verilator!

```
[error] <redacted>/chipyard/generators/chipyard/src/main/scala/config/GemminiSoCConfigs.scala:13:11: parameter 'nWays' is already specified at parameter position 1
[error] Note that 'nBanks' is not a parameter name of the invoked method.
[error]     nWays = 8,
[error]           ^
[error] one error found
[error] (Compile / compileIncremental) Compilation failed
```

jerryz123, 2022-10-07:
0.6.4 is an older release that is not compatible with 1.8.0. 

More specifically, the CustomSoCConfigs file in Gemmini should be updated to reflect API changes in 1.8.0.

hngenc, 2022-10-07:
For Gemmini 0.6.4, you should be using Chipyard's `117624d8eea27bafd613eec09e9b9b3e31239e08` commit (as in the `CHIPYARD.hash` file)

Was there as specific reason why you wanted Chipyard 1.8.0? If so, one easy way to use it might by to just remove the `nBanks` option from `CustomSoCConfigs.scala`, since the error message implies that the option no longer exists in newer versions of Chipyard

hanyax, 2022-11-15:
Hey, I am also trying to build this on Chipyard/FireSim, looks like there is a compatibility issue where I need Chipyard 18.0 and FireSim 14.2 to run on the new AWS FPGA AMI but Gemmini does not support Chipyard 18.0?

isbord, 2022-11-24:
Hi,

If this can help, I did similar test with Chipyard 1.8.1 and Gemmini 0.6.4 and, of course, had exactly the same problem.
I therefore commented lines 12 & 13 of `generators/chipyard/src/main/scala/config/GemminiSoCConfigs.scala`:
```
 10   new freechips.rocketchip.subsystem.WithInclusiveCache(
 11     //nBanks = 1,
 12     //nWays = 8,
 13     capacityKB = 512,
 14     outerLatencyCycles = 40
 15   ) ++
```

and now, I successfully pass the mobilenet test (`./scripts/run-verilator.sh mobilenet`)

hngenc, 2022-11-28:
The latest release of Gemmini on the `master` branch (tagged `v0.7.0`) is compatible with Chipyard 1.8. Hope that fixes your problems