ucb-bar/gemmini #249: gemmini custom config  building failed

chithize, 2022-10-14:
hi  

Tried to change gemmini custom config to high performance config, but failed with the below error message:
root@nb-cs1:~/chipyard/generators/gemmini# ./scripts/build-verilator.sh -j8
Running with RISCV=/root/chipyard/esp-tools-install
mkdir -p /root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig
cd /root/chipyard && java -jar /root/chipyard/generators/rocket-chip/sbt-launch.jar   ";project chipyard; runMain chipyard.Generator  --target-dir /root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig --name chipyard.TestHarness.CustomGemminiSoCConfig --top-module chipyard.TestHarness --legacy-configs chipyard:CustomGemminiSoCConfig "
Picked up JAVA_TOOL_OPTIONS: -Xmx8G -Xss8M -Djava.io.tmpdir=/root/chipyard/.java_tmp
[info] welcome to sbt 1.5.5 (Ubuntu Java 11.0.16)
[info] loading settings for project chipyard-build from plugins.sbt ...
[info] loading project definition from /root/chipyard/project
[info] loading settings for project chipyardRoot from build.sbt ...
[info] loading settings for project hardfloat from build.sbt ...
[info] loading settings for project rocketConfig from build.sbt ...
[info] loading settings for project testchipip from build.sbt ...
[info] loading settings for project icenet from build.sbt ...
[info] loading settings for project hwacha from build.sbt ...
[info] loading settings for project boom from build.sbt ...
[info] loading settings for project cva6 from build.sbt ...
[info] loading settings for project ibex from build.sbt ...
[info] loading settings for project sodor from build.sbt ...
[info] loading settings for project gemmini from build.sbt ...
[info] loading settings for project tapeout from build.sbt ...
[info] loading settings for project sim-build from plugins.sbt ...
[info] loading project definition from /root/chipyard/sims/firesim/sim/project
[info] loading settings for project firesim from build.sbt ...
[info] loading settings for project targetutils from build.sbt ...
[info] loading settings for project midas from build.sbt ...
[info] resolving key references (35265 settings) ...
[info] set current project to chipyardRoot (in build file:/root/chipyard/)
[info] set current project to chipyard (in build file:/root/chipyard/)
[info] running chipyard.Generator --target-dir /root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig --name chipyard.TestHarness.CustomGemminiSoCConfig --top-module chipyard.TestHarness --legacy-configs chipyard:CustomGemminiSoCConfig
(3,List(UInt<7>(123)))

Exception: sbt.TrapExitSecurityException thrown from the UncaughtExceptionHandler in thread "run-main-0"
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 8 s, completed Oct 14, 2022, 7:49:31 AM
/root/chipyard/common.mk:110: recipe for target 'generator_temp' failed
make: *** [generator_temp] Error 1

some one can help to resolve it?  what is the problem?

hngenc, 2022-11-23:
Hmm, I can't seem to replicate that when I build the `highPerfInferenceConfig`. Can you share the Gemmini commit and Chipyard commit that you were on when you got that error?

It's also possible that you just run out of RAM when building. That often causes errors with unhelpful error messages.

chithize, 2022-11-27:
Hasan 

Thank for the reply,  I skip this problem with tutorial docker image. please omit it