ucb-bar/gemmini #250: no such file or directory: toolchains/esp-tools/riscv-isa-sim/build

linuxlonelyeagle, 2022-10-20:
When i use **./scripts/build-toolchains.sh esp-tools**, the is an error
```
cd toolchains/esp-tools/riscv-isa-sim/build
cd: no such file or directory: toolchains/esp-tools/riscv-isa-sim/build
```
I have spent a long time trying to solve this problem, but have not found out what the problem is.

linuxlonelyeagle, 2022-10-21:
I found that it finally outputs **Nothing to be done for 'install-data-am'.** If anyone can help me, I would be grateful.
```
make[12]: Nothing to be done for 'install-data-am'.
make[12]: Leaving directory '/home/sen/chipyard/toolchains/esp-tools/riscv-gnu-toolchain/build/build-gdb-newlib/gdb/build-gnulib/import'
make[11]: Leaving directory '/home/sen/chipyard/toolchains/esp-tools/riscv-gnu-toolchain/build/build-gdb-newlib/gdb/build-gnulib/import'
make[10]: Leaving directory '/home/sen/chipyard/toolchains/esp-tools/riscv-gnu-toolchain/build/build-gdb-newlib/gdb/build-gnulib/import'
make[9]: Leaving directory '/home/sen/chipyard/toolchains/esp-tools/riscv-gnu-toolchain/build/build-gdb-newlib/gdb/build-gnulib/import'
make[8]: Leaving directory '/home/sen/chipyard/toolchains/esp-tools/riscv-gnu-toolchain/build/build-gdb-newlib/gdb/build-gnulib'
make[7]: Leaving directory '/home/sen/chipyard/toolchains/esp-tools/riscv-gnu-toolchain/build/build-gdb-newlib/gdb/build-gnulib'
make[6]: Leaving directory '/home/sen/chipyard/toolchains/esp-tools/riscv-gnu-toolchain/build/build-gdb-newlib/gdb/build-gnulib'
make[5]: Leaving directory '/home/sen/chipyard/toolchains/esp-tools/riscv-gnu-toolchain/build/build-gdb-newlib/gdb'
make[4]: Leaving directory '/home/sen/chipyard/toolchains/esp-tools/riscv-gnu-toolchain/build/build-gdb-newlib/gdb'
make[3]: Leaving directory '/home/sen/chipyard/toolchains/esp-tools/riscv-gnu-toolchain/build/build-gdb-newlib/gdb'
make[2]: Leaving directory '/home/sen/chipyard/toolchains/esp-tools/riscv-gnu-toolchain/build/build-gdb-newlib'
make[1]: Leaving directory '/home/sen/chipyard/toolchains/esp-tools/riscv-gnu-toolchain/build/build-gdb-newlib'
```

linuxlonelyeagle, 2022-10-21:
But when i use the command 
```
cd toolchains/esp-tools/riscv-isa-sim/build
make && make install 
```
there is an error 
```
Makefile:349: warning: overriding recipe for target 'disasm.o'
Makefile:349: warning: ignoring old recipe for target 'disasm.o'
make: Circular libhwacha.so <- libhwacha.so dependency dropped.
make: Circular libgemmini.so <- libgemmini.so dependency dropped.
make: Circular libsha3.so <- libsha3.so dependency dropped.
make: Circular libcustomext.so <- libcustomext.so dependency dropped.
make: Circular libsoftfloat.so <- libsoftfloat.so dependency dropped.
make: Nothing to be done for 'default'.
Makefile:349: warning: overriding recipe for target 'disasm.o'
Makefile:349: warning: ignoring old recipe for target 'disasm.o'
../scripts/mk-install-dirs.sh /home/sen/chipyard/esp-tools-install/include
for file in fesvr/byteorder.h fesvr/elf.h fesvr/elfloader.h fesvr/htif.h fesvr/dtm.h fesvr/memif.h fesvr/syscall.h fesvr/context.h fesvr/htif_pthread.h fesvr/htif_hexwriter.h fesvr/option_parser.h fesvr/term.h fesvr/device.h fesvr/rfb.h fesvr/tsi.h riscv/mmio_plugin.h; \
  do \
        ../scripts/mk-install-dirs.sh /home/sen/chipyard/esp-tools-install/include/`dirname $file`; \
    /usr/bin/install -c -m 644 ../$file /home/sen/chipyard/esp-tools-install/include/`dirname $file`; \
  done
../scripts/mk-install-dirs.sh /home/sen/chipyard/esp-tools-install/include
for dir in fesvr          ; \
  do \
        ../scripts/mk-install-dirs.sh /home/sen/chipyard/esp-tools-install/include/$dir; \
    /usr/bin/install -c -m 644 config.h /home/sen/chipyard/esp-tools-install/include/$dir; \
  done
make: Circular libhwacha.so <- libhwacha.so dependency dropped.
make: Circular libgemmini.so <- libgemmini.so dependency dropped.
make: Circular libsha3.so <- libsha3.so dependency dropped.
make: Circular libcustomext.so <- libcustomext.so dependency dropped.
make: Circular libsoftfloat.so <- libsoftfloat.so dependency dropped.
../scripts/mk-install-dirs.sh /home/sen/chipyard/esp-tools-install/lib
for file in libfesvr.a libhwacha.so libgemmini.so libsha3.so libdisasm.a libcustomext.so libsoftfloat.so; \
  do \
    /usr/bin/install -c -m 644 $file /home/sen/chipyard/esp-tools-install/lib; \
  done
../scripts/mk-install-dirs.sh /home/sen/chipyard/esp-tools-install/bin
for file in elf2hex spike spike-log-parser xspike termios-xspike spike-dasm; \
  do \
    /usr/bin/install -c -m 755 $file /home/sen/chipyard/esp-tools-install/bin; \
  done
../scripts/mk-install-dirs.sh /home/sen/chipyard/esp-tools-install/lib/pkgconfig/
for file in riscv-fesvr.pc riscv-disasm.pc; \
  do \
    /usr/bin/install -c -m 644 $file /home/sen/chipyard/esp-tools-install/lib/pkgconfig/; \
  done

```

hngenc, 2022-11-23:
Try running `source env.sh` in `chipyard/`. That's usually enough to fix this error.

linuxlonelyeagle, 2022-11-24:
> Try running `source env.sh` in `chipyard/`. That's usually enough to fix this error.

I think the main problem is that the dependencies are not installed in the system.I probably know what `source env.sh` does.[Setting up the Chipyard Rope](https://chipyard.readthedocs.io/en/stable/Chipyard-Basics/Initial-Repo-Setup.html#prerequisites) does not say which packages need to be installed.

hngenc, 2022-11-24:
Try these dependencies:

https://chipyard.readthedocs.io/en/1.7.1/Chipyard-Basics/Initial-Repo-Setup.html#requirements

linuxlonelyeagle, 2022-11-25:
> Try these dependencies:
> 
> https://chipyard.readthedocs.io/en/1.7.1/Chipyard-Basics/Initial-Repo-Setup.html#requirements

Thanks! In the latest documentation, I don't see these dependencies, but the problem has been resolved.
Sorry, I want to ask some risc-v isa questions. I'm interested in computer architecture and compilers, I want to emulate similar instructions using LLVM, what risc-v extension is gemmini using? Is gemmini using standard extension? Are the risc-v instructions here 128bit ? I'm not very clear, is there any relevant risc-v material that I can learn? I'm sorry that the question I asked was not about gemmini.

hngenc, 2022-11-25:
Gemmini is using RoCC, which is a non-standard custom RISCV 64-bit extension. You can read more about it here:

https://chipyard.readthedocs.io/en/stable/Customization/RoCC-or-MMIO.html

linuxlonelyeagle, 2022-11-26:
> Gemmini is using RoCC, which is a non-standard custom RISCV 64-bit extension. You can read more about it here:
> 
> https://chipyard.readthedocs.io/en/stable/Customization/RoCC-or-MMIO.html

Oh!I actually read this document before.But there is one thing that confuses me.
`mvin` Move Data From Main Memory to Scratchpad
Format: mvin rs1, rs2

rs1 = virtual DRAM address (byte addressed) to load into scratchpad
rs2[31:0] = local scratchpad or accumulator address
rs2[47:32] = number of columns to load in
rs2[63:48] = number of rows to load in. Must be less than or equal to DIM.
funct = 2

Why is r2 a 64-bit immediate number? An instruction has only 64 bits and the rest of the instruction cannot be represented at all. Or is r2 stored in a register?? Here we are using the register where r2 is stored.

hngenc, 2022-11-26:
`rs1` and `rs2` are both 64-bit registers

linuxlonelyeagle, 2022-11-26:
> `rs1` and `rs2` are both 64-bit registers

It's a pleasure to ask you questions.Thanks!

hngenc, 2022-11-26:
No problem!