ucb-bar/gemmini #251: perl: warning: Setting locale failed.

linuxlonelyeagle, 2022-10-21:
I am having a problem with what I assume is a character encoding problem, but I am not sure. If anyone can help me, I would be grateful.
```
[error] Picked up JAVA_TOOL_OPTIONS: -Xmx8G -Xss8M -Djava.io.tmpdir=/home/sen/chipyard/.java_tmp
[success] Total time: 3 s, completed Oct 22, 2022, 1:11:43 AM
sort -u /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/sim_files.f /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/firrtl_black_box_resource_files.top.f /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/firrtl_black_box_resource_files.harness.f | grep -v '.*\.\(svh\|h\)$' > /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/sim_files.common.f
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
rm -rf /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig
mkdir -p /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig
verilator --cc --exe -CFLAGS " -O3 -std=c++11 -I/home/sen/chipyard/esp-tools-install/include -I/home/sen/chipyard/tools/DRAMSim2 -I/home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig    -D__STDC_FORMAT_MACROS -DTEST_HARNESS=VTestHarness -DVERILATOR -include /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.plusArgs -include /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/verilator.h" -LDFLAGS " -L/home/sen/chipyard/esp-tools-install/lib -Wl,-rpath,/home/sen/chipyard/esp-tools-install/lib -L/home/sen/chipyard/sims/verilator -L/home/sen/chipyard/tools/DRAMSim2 -lfesvr -ldramsim "   --threads 1 --threads-dpi all -O3 --x-assign fast --x-initial fast --output-split 10000 --output-split-cfuncs 100 --assert -Wno-fatal --timescale 1ns/1ps --max-num-width 1048576 +define+PRINTF_COND=\$c\(\"verbose\",\"\&\&\"\,\"done_reset\"\) +define+STOP_COND=\$c\(\"done_reset\"\) --top-module TestHarness --vpi -f /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/sim_files.common.f /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.v /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.harness.v /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.mems.v /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.harness.mems.v  -o /home/sen/chipyard/sims/verilator/simulator-chipyard-CustomGemminiSoCConfig -Mdir /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig -CFLAGS "-include /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig/VTestHarness.h"
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
%Warning-WIDTH: /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.v:219747:70: Operator DIV expects 9 bits on the RHS, but RHS's CONST '7'h40' generates 7 bits.
                                                                                                                                                                       : ... In instance TestHarness.chiptop.system.tile_prci_domain.tile_reset_domain.tile.gemmini.spad.reader.beatPacker
219747 |   wire [8:0] _io_out_bits_addr_T = io_out_bits_addr_total_bytes_sent / 7'h40;  
       |                                                                      ^
                ... For warning description see https://verilator.org/warn/WIDTH?v=5.001
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
%Warning-WIDTH: /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.v:219748:72: Operator DIV expects 9 bits on the RHS, but RHS's CONST '5'h10' generates 5 bits.
                                                                                                                                                                       : ... In instance TestHarness.chiptop.system.tile_prci_domain.tile_reset_domain.tile.gemmini.spad.reader.beatPacker
219748 |   wire [8:0] _io_out_bits_addr_T_1 = io_out_bits_addr_total_bytes_sent / 5'h10;  
       |                                                                        ^
%Warning-WIDTH: /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.v:216618:56: Operator DIV expects 6 bits on the RHS, but RHS's CONST '5'h10' generates 5 bits.
                                                                                                                                                                       : ... In instance TestHarness.chiptop.system.tile_prci_domain.tile_reset_domain.tile.gemmini.spad.reader.core
216618 |   wire [5:0] _io_reserve_entry_addr_T = bytesRequested / 5'h10;  
       |                                                        ^
%Warning-WIDTH: /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.v:301160:34: Operator DIV expects 16 bits on the RHS, but RHS's CONST '5'h10' generates 5 bits.
                                                                                                                                                                       : ... In instance TestHarness.chiptop.system.tile_prci_domain.tile_reset_domain.tile.gemmini.mod.st_
301160 |   wire [15:0] _spad_addr_T = och / 5'h10;  
       |                                  ^
%Warning-WIDTH: /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.v:300321:34: Operator DIV expects 16 bits on the RHS, but RHS's CONST '5'h10' generates 5 bits.
                                                                                                                                                                       : ... In instance TestHarness.chiptop.system.tile_prci_domain.tile_reset_domain.tile.gemmini.mod.ex_
300321 |   wire [15:0] _a_addr_T_14 = kch / 5'h10;  
       |                                  ^
%Warning-WIDTH: /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.v:300330:31: Operator DIV expects 16 bits on the RHS, but RHS's CONST '5'h10' generates 5 bits.
                                                                                                                                                                       : ... In instance TestHarness.chiptop.system.tile_prci_domain.tile_reset_domain.tile.gemmini.mod.ex_
300330 |   wire [15:0] _c_addr_T = och / 5'h10;  
       |                               ^
%Warning-WIDTH: /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.top.v:299643:37: Operator DIV expects 16 bits on the RHS, but RHS's CONST '5'h10' generates 5 bits.
[... cut ...]
%Error-NEEDTIMINGOPT: /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/SimJTAG.v:59:11: Use --timing or --no-timing to specify how timing controls should be handled
                                                                                                                                 : ... In instance TestHarness.SimJTAG
   59 |    assign #0.1 exit = __exit;
      |           ^
%Error: Exiting due to 5 error(s)
make: *** [Makefile:216: /home/sen/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig/VTestHarness.mk] Error 1
```

linuxlonelyeagle, 2022-10-26:
@hngenc  Hi!Can you help me?I really need your help.This question has been bothering me for a long time.I would be grateful if you could help me.Thanks!

hngenc, 2022-11-23:
So sorry for the late response. I haven't seen this error before, to be honest. This doesn't seem to be failing due to a locale issue, but due to the `NEEDTIMINGOPT` error instead (which I haven't run into before).

Can you share the Gemmini commit and Chipyard commit you were on when you encountered this? I'll try to reproduce the error.

linuxlonelyeagle, 2022-11-24:
> So sorry for the late response. I haven't seen this error before, to be honest. This doesn't seem to be failing due to a locale issue, but due to the `NEEDTIMINGOPT` error instead (which I haven't [...]
Thank you for your reply.This is the commit for chipyard.
```
commit 117624d8eea27bafd613eec09e9b9b3e31239e08 (HEAD)
Author: abejgonzalez <abe.j.gonza@gmail.com>
Date:   Mon Jan 17 16:23:47 2022 -0800

    Bump BOOM/Gemmini

commit cd0d194ba63f5f148fb5c73a2a3b6bc354b912fc
Author: abejgonzalez <abe.j.gonza@gmail.com>
Date:   Mon Jan 17 16:12:22 2022 -0800

    Update rocketchip

commit 9845657a070d9acceff11f0c9de402bee2b1a157
Author: abejgonzalez <abe.j.gonza@gmail.com>
Date:   Mon Jan 17 15:52:26 2022 -0800

    Bump FireSim

commit 12634276559eb4a722c4ad625a0a8772c3e70a30
Author: abejgonzalez <abe.j.gonza@gmail.com>
Date:   Mon Jan 17 15:46:37 2022 -0800

    Allow unrecognized annos

commit 70202eb9df4cd14370b757a77da0953bcb6d5831
Author: abejgonzalez <abe.j.gonza@gmail.com>
Date:   Mon Jan 17 15:19:00 2022 -0800
...
```
This is the Gemmini commit.
```
commit 87550581d98210540ac013f96fbcb9150be39df7 (HEAD, tag: v0.6.4)
Author: Hasan Genc <hngenc@berkeley.edu>
Date:   Mon Mar 7 11:27:05 2022 -0800

    Update version number in Quick Setup instructions

commit b9eda0567d6b53a514a64367745977d8561567b8
Merge: c47cb7f 5b8e24c
Author: Hasan Genc <hngenc@berkeley.edu>
Date:   Mon Mar 7 11:26:06 2022 -0800

    Merge pull request #205 from ucb-bar/dev
    
    Bump Gemmini to v0.6.4
    
    New features:
    * Add support for new "dummy configs" which give cycle-accurate performance results, but which don't guarantee functional correctness of matmul/conv operations
    * Optimize area for reservation stations by splitting unified reservation station into separate stations for each instruction type
    
    Bug fixes:
    * Fix incorrect assertions from firing when building some non-systolic configs
    * Fix unpredictable infinite stall caused by OS interrupts
    * Fix build scripts when building simulators that can generate waveforms
   ... 
```

hngenc, 2022-11-26:
Those commits should be working fine; I'm not sure why they failed for you.

Can you also post the output of your `locale` command? I'm not sure if your locale is the error, but it shouldn't hurt to check.

linuxlonelyeagle, 2022-11-26:
> Those commits should be working fine; I'm not sure why they failed for you.
> 
> Can you also post the output of your `locale` command? I'm not sure if your locale is the error, but it shouldn't hurt to check.
```
❯ locale
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="C"
LC_NUMERIC="C"
LC_TIME="C"
LC_COLLATE="C"
LC_MONETARY="C"
LC_MESSAGES="C"
LC_PAPER="C"
LC_NAME="C"
LC_ADDRESS="C"
LC_TELEPHONE="C"
LC_MEASUREMENT="C"
LC_IDENTIFICATION="C"
LC_ALL=C
```

hngenc, 2022-11-26:
How about you try running `export LC_ALL=en_US.UTF-8`, and then running the `build-verilator.sh` script again?

linuxlonelyeagle, 2022-11-26:
> How about you try running `export LC_ALL=en_US.UTF-8`, and then running the `build-verilator.sh` script again?

```
❯ export LC_ALL=en_US.UTF-8
❯ locale
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=en_US.UTF-8
```
Thanks!I think it's because something is wrong here and I will fix it.Then running the `build-verilator.sh`.

linuxlonelyeagle, 2022-11-26:
```
❯ locale
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=en_US.UTF-8
```
But when i running the `build-verilator.sh`,the error reappears.

hngenc, 2022-11-28:
Can you try doing this with the latest version of Gemmini (`v0.7.0`) on the `master` branch? Note that it also requires a different Chipyard version than the one you're using now: https://github.com/ucb-bar/gemmini#quick-start

I'm hoping that using a more recent Chipyard version fixes your bug

linuxlonelyeagle, 2022-11-29:
> Can you try doing this with the latest version of Gemmini (`v0.7.0`) on the `master` branch? Note that it also requires a different Chipyard version than the one you're using now: https://github.com/ucb-bar/gemmini#quick-start
> 
> I'm hoping that using a more recent Chipyard version fixes your bug

This problem has been solved!Thank you for everything you do.