ucb-bar/gemmini #253: Size of the systolic array

uditagarwal97, 2022-11-07:
Hi,

I recently tried to synthesize a systolic array of size 128x128 using FireSim and EC2 F1 instances, however, I got an error saying that the number of system logic cells required to map my design to the FPGA is a lot more than the ones available on the FPGA (Resource overuse errors).

I just wanted to check-in with you guys about what's the maximum size of the systolic array in Gemmini have you experimented with?

gnodipac886, 2022-11-23:
@uditagarwal97 I wanted to test this too, how did you happen to modify the size of the systolic array?

hngenc, 2022-11-23:
We've built a 32-by-32 design on Firesim a while back, but I don't remember us pushing above that. Did 64-by-64 work for you?

Also, I believe Firesim gives multiple options for FPGAs. Perhaps there are larger FPGAs you can try instead?