ucb-bar/gemmini #254: errors while building verilator

Charles-Tang, 2022-11-10:
When I run ./scripts/build-verilator.sh I encontered errors listed below, any idea how I can solve this?:

%Error-NEEDTIMINGOPT: /media/charles/0F6215370F621537/Files/BachelorDesign/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/SimJTAG.v:54:11: Use --timing or --no-timing to specify how timing controls should be handled
                                                                                                                                                                            : ... In instance TestHarness.SimJTAG
   54 |    assign #0.1 jtag_TCK   = __jtag_TCK;
      |           ^
%Error-NEEDTIMINGOPT: /media/charles/0F6215370F621537/Files/BachelorDesign/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/SimJTAG.v:55:11: Use --timing or --no-timing to specify how timing controls should be handled
                                                                                                                                                                            : ... In instance TestHarness.SimJTAG
   55 |    assign #0.1 jtag_TMS   = __jtag_TMS;
      |           ^
%Error-NEEDTIMINGOPT: /media/charles/0F6215370F621537/Files/BachelorDesign/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/SimJTAG.v:56:11: Use --timing or --no-timing to specify how timing controls should be handled
                                                                                                                                                                            : ... In instance TestHarness.SimJTAG
   56 |    assign #0.1 jtag_TDI   = __jtag_TDI;
      |           ^
%Error-NEEDTIMINGOPT: /media/charles/0F6215370F621537/Files/BachelorDesign/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/SimJTAG.v:57:11: Use --timing or --no-timing to specify how timing controls should be handled
                                                                                                                                                                            : ... In instance TestHarness.SimJTAG
   57 |    assign #0.1 jtag_TRSTn = __jtag_TRSTn;
      |           ^
%Error-NEEDTIMINGOPT: /media/charles/0F6215370F621537/Files/BachelorDesign/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/SimJTAG.v:59:11: Use --timing or --no-timing to specify how timing controls should be handled
                                                                                                                                                                            : ... In instance TestHarness.SimJTAG
   59 |    assign #0.1 exit = __exit;
      |           ^
%Error: Exiting due to 5 error(s)
make: *** [Makefile:222：/media/charles/0F6215370F621537/Files/BachelorDesign/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/chipyard.TestHarness.CustomGemminiSoCConfig.debug/VTestHarness.mk] 错误 1

hngenc, 2022-11-23:
Seems to be related to #251.

Can you share your Gemmini commit and Chipyard commit? I'll try to reproduce this error.

hngenc, 2022-11-29:
#251 was fixed in the latest Gemmini release (`v0.7.0`), so I'm closing this issue for now. But feel free to reopen it if you still run into this problem.

closed 2022-11-29 by hngenc