ucb-bar/gemmini #255: [error] import midas.targetutils.PerfCounter

e665107, 2022-11-17:
Hi
I konw it depends on the midas, But i don't konw how to add it,  Can you give me some solustions? 
Thanks!

hngenc, 2022-11-23:
It should be included for you by Firesim. Can you share which Gemmini commit and which Chipyard commit you're on?

e665107, 2022-11-24:
Thank you for your reply!  I am working on the following prj:
https://github.com/eugene-tarassov/vivado-risc-v
I follow the operations of the readme to use it. But my net doesn't work well, so i have no choice but cloning the relate 
lib independently. And go on. When it compiles ,  it indicates the "import midas.targetutils.PerfCounter" error.

hngenc, 2022-11-24:
Do you mean you can't clone all of Chipyard at once? We don't really support "partial cloning", unfortunately. I'm sure it's possible, but I just haven't tried it out myself.

If the PerfCounters are giving you a lot of trouble, it might be easier for you to just delete all the lines where they appear in Gemmini. They're only used for Firesim profiling anyways.

e665107, 2022-11-24:
Believe it or not, when i clone the relate lib solely, it works. However, when i use the git submodule --init  --recursive, it fails. Clone nothing down!

hngenc, 2022-11-24:
Thanks for checking that! That's surprising to me

e665107, 2022-11-25:
[success] Total time: 15 s, completed Nov 25, 2022, 12:48:17 PM
rm workspace/bootrom.img
java -Xmx12G -Xss8M  -cp target/scala-2.12/classes:rocket-chip/rocketchip.jar firrtl.stage.FirrtlMain -i workspace/rocket32s1/system-kc705/Vivado.Rocket32s1.fir -o system-kc705.v -X verilog --infer-rw RocketSystem --repl-seq-mem \
-c:RocketSystem:-o:`realpath workspace/rocket32s1/system.conf` \
-faf `realpath workspace/rocket32s1/system.anno.json` \
-td workspace/rocket32s1/ \
-fct firrtl.passes.InlineInstances
Killed
make: *** [Makefile:221: workspace/rocket32s1/system-kc705.v] Error 137

Hi, thanks for your last reply.
The log above suggest me that, it needs firrtl.  Do you konw which directory firrtl should in? Or, How can i use it correctly?
Now, i have compiled the firrtl successfully, and place the utils/bin/firrtl to my PATH. But it doesn't work. Do you konw where I am 
wrong?

hngenc, 2022-11-25:
I'm not sure, unfortunately. We just haven't tried cloning the modules individually before, so I don't know what path or dependency issues might result from that