ucb-bar/gemmini #263: Invalid pointer error using WS Gemmini

mbelda, 2022-12-02:
Hello,

Do you know if there is any bug with the WS execution for Gemmini?

I am trying to run a cnn using the example workload from `onnxruntime-riscv/systolic_runner/imagenet_runner` barely modified for my network (just labels and preprocessing) and I am getting an error with the WS option.

My current configuration for Gemmini is as follows: 
~~~
//FP32 Single Precision Configuration
  val FP32Mesh8WithConvsHighPerfConfig = defaultFPConfig.copy(
    inputType = Float(8, 24),
    spatialArrayOutputType = Float(8, 24),
    accType = Float(8, 24),
    tile_latency = 2,
    has_training_convs = true,
    meshRows = 8,
    meshColumns = 8,
    ex_read_from_acc = false,
    ex_write_to_spad = false,
    mvin_scale_args = Some(ScaleArguments((t: Float, u: Float) => t * u, 4, Float(8, 24), -1, identity = "1.0", c_str="((x) * (scale))")),
    mvin_scale_acc_args = Some(ScaleArguments((t: Float, u: Float) => t * u, 4, Float(8, 24), -1, identity = "1.0", c_str="((x) * (scale))")),
  )
~~~

As I have enabled convs the OS mode is not available, so I need to execute with the WS option but it throws an error.
I have tried to run it with the CPU option and it works fine.

The following error is shown on the output:
~~~
Mobilenet en Gemmini WS
Loaded runner program
Finished mlockall
Using systolic in mode 2
Using Onnxruntime C++ API
Number of inputs = 1
Input 0 : name=input, type=1, num_dims=4: [1, 224, 224, 3, ]
Number of outputs = 1
Output 0 : name=dense_2, type=1, num_dims=2: [1, 13, ]
Loading image
Image dimensions: 224 224 3
First few image values -0.600000 -0.725490 -0.843137
Everything finde over here
Lets run the model
free(): invalid pointer
/firesim.sh: line 13:   231 Aborted                 ./ort_test -m MobileNetV2_0p5_all.onnx -i images/test_P_3.jpg -p "mobchess" -x 2 -O 99
~~~

I narrowed down the error `free():invalid pointer` to this line of my code:
~~~
auto output_tensors = session.Run(Ort::RunOptions{nullptr}, input_node_names.data(), &input_tensor, 1, output_node_names.data(), 1);
~~~

But it works fine for CPU execution so I am guessing it has something to do with Gemmini.

I attach the binary, the network and one image in case you need them.
https://drive.google.com/file/d/1--kqAZ4LEHu_OJO20qMLB10AsB9Gi7Gd/view?usp=sharing
(It is a link for google drive since github does not let me upload large files)

Sorry to bother you again.
Best regards, MªJosé.

hngenc, 2022-12-02:
1. Usually, memory errors are caused by a change in `DIM`, `inputType` or `accType`, rather than by changing the dataflow. Can you double-check that your `systolic_params_*.h` files have `DIM 8`, `inputType float`, and `accType float`?
2. Sometimes, after changing the Gemmini configs, you need to run:

```
cd ./software/onnxruntime-riscv/
rm -rf ./build/
```

before rebuilding ONNX-Runtime. Can you try doing that as well, and seeing if you still get this error?
3. Do you know if this error happens before or after you've made any call into `tiled_conv_auto`, `tiled_matmul_auto`, or `tiled_resadd_auto`? Can you search for where they're called in ONNX-Runtime and include `printf`s before and after they're called? It's possible that one of these functions is reading/writing out-of-bounds, which could cause memory-corruption errors.

mbelda, 2022-12-02:
Rebuilding the onnxruntime-riscv did it.
I finally got good results.
Thanks!

closed 2022-12-02 by mbelda

hngenc, 2022-12-02:
That's great; thanks for confirming!