ucb-bar/gemmini #264: Is there a tutorial for fpga here?

linuxlonelyeagle, 2022-12-04:
I want to run gemmini on fpga.Is there a tutorial for fpga here?Thanks!

gnodipac886, 2022-12-04:
@linuxlonelyeagle you can checkout this repo here :)
https://github.com/eugene-tarassov/vivado-risc-v

linuxlonelyeagle, 2022-12-05:
> @linuxlonelyeagle you can checkout this repo here :) https://github.com/eugene-tarassov/vivado-risc-v

Thanks!

hngenc, 2022-12-05:
We haven't yet attempted to run Gemmini on a standalone FPGA, although I know some people have managed to do so. But it's not a workflow that we officially support.

However, I believe that more recent versions of Firesim have experimental support for on-premise FPGAs. So you might be able to run Gemmini workloads on your personal FPGAs through Firesim.

linuxlonelyeagle, 2022-12-06:
> We haven't yet attempted to run Gemmini on a standalone FPGA, although I know some people have managed to do so. But it's not a workflow that we officially support.
>
> However, I believe that more [...]

Thanks!

closed 2022-12-06 by linuxlonelyeagle