ucb-bar/gemmini #266: im2col feature on gemmini

mbelda, 2022-12-19:
Hello,

When I read the Gemmini paper (https://people.eecs.berkeley.edu/~ysshao/assets/papers/genc2021-dac.pdf) a im2col feature is mentioned.

This feature is supposed to be on the Rocket and Boom cores and also on the accelerator.
And it seems to be very important when executing neural networks.

Is enabled by default? My speed times are congruent with the feature disabled on Gemmini and I would like to have the results with the tim2col on gemmini.

Thank your very much!

Best regards, MªJosé.

hngenc, 2022-12-22:
The im2col features are enabled by default now.

Are you sure your bottleneck is actually due to im2col? There might be some other operation that's slowing you down. If you're running ONNX-Runtime, you can use the `-t` option to specify a trace file that can help identify bottlenecks. Could you create a trace file and share it?

messiic, 2023-05-10:
Hi,
I also has a question about im2col issue. I want to know if these is a software case using im2col function? Actually, I just doubt that if im2col can take effect in current Gemmini code. I find that im2col_en is enabled only when wegiht_stride!=0, and weight_stride is overwrited by cmd.bits(0).cmd.rs2(22, 20) when config_cmd_type!==CONFIG_EX. But only when config_cmd_type is euqal to CONFIG_EX, the config cmd is passed to ExecuteController in my mind. 
So, Looking forward to your reply. Thank your very much!
Best regards, Yuexin Zhao

hngenc, 2023-05-22:
The `im2col_en` signal isn't being used anymore. (We'll get rid of that signal in the future; we just haven't gotten around to it yet).

What the im2col modules were previously doing is has basically been taken over by the `LoopConv` module. However, you can still perform im2col on the CPU and run the resulting matmul on Gemmini's `LoopMatmul` module if you want to compare the two.

messiic, 2023-05-26:
@hngenc Thans you for your detailed answer！I will study more on LoopConv about im2col feature.