ucb-bar/gemmini #268: Where can I find the simulation logs if I run Spike and Verilator on my computer?

hyz9181, 2022-12-19:
Hi. After I finish running gemmini ./scripts/run-spike.sh or ./scripts/run-verilator.sh on my personal computer, where can I find the logs of spike or verilator simulation log?
Thank you.

hngenc, 2022-12-22:
What kind of simulation logs are you thinking about? If you run `run-spike.sh` or `run-verilator.sh`, the program's output will be printed to `stdout`.

hyz9181, 2022-12-22:
I see. I tried the run-spike.sh and it works fine but when I runt he run-verilator.sh with resnet50, mobilenet, or mlp, seems like the verilator is not running. Any idea for this one? I think I have followed the instruction to setup the chipyard and gemmini.

hngenc, 2022-12-22:
resnet50, mobilenet, and mlp are very long-running tests. We would
recommend running them on Firesim, but if you run them on Verilator, they
will probably take several days to run.