ucb-bar/gemmini #272: terminate called after throwing an instance of 'trap_load_access_fault'

linuxlonelyeagle, 2023-01-06:
I extended the backend of llvm, then used llc to generate the obj file, then used gcc to generate the executable, but I had problems running the spike.There is my man function.
```
0000000000010198 <main>:
   10198:       00100513                li      a0,1
   1019c:       0e05307b                custom3 0,a0,zero,7
   101a0:       00008067                ret
```
The 0e05307b is **0000111 00000 01010 011 00000 1111011**.I disassembled the **mvin_mvout-baremetal**.
```
00000000800017b6 <main>:
    800017b6:   711d                    addi    sp,sp,-96
    800017b8:   ec86                    sd      ra,88(sp)
    800017ba:   e8a2                    sd      s0,80(sp)
    800017bc:   e4a6                    sd      s1,72(sp)
    800017be:   e0ca                    sd      s2,64(sp)
    800017c0:   fc4e                    sd      s3,56(sp)
    800017c2:   f852                    sd      s4,48(sp)
    800017c4:   f456                    sd      s5,40(sp)
    800017c6:   f05a                    sd      s6,32(sp)
    800017c8:   ec5e                    sd      s7,24(sp)
    800017ca:   e862                    sd      s8,16(sp)
    800017cc:   e466                    sd      s9,8(sp)
    800017ce:   4781                    li      a5,0
    800017d0:   0ef7b07b                custom3 0,a5,a5,7
```
The 0ef7b07b is **0000111 01111 01111 011 00000 1111011**,but when I run spike,it show that
```
 $:spike --extension=gemmini a.out
Gemmini extension configured with:
    dim = 16
terminate called after throwing an instance of 'trap_load_access_fault'
```
I had checked the https://github.com/ucb-bar/gemmini#flush-flushes-the-tlb. If someone helps me, I would be grateful.Thanks!

linuxlonelyeagle, 2023-01-09:
This problem has been solved.

closed 2023-01-09 by linuxlonelyeagle

HVwang123, 2023-11-17:
Can you tell me how you solved the problem? The same error occurred to me.

linuxlonelyeagle, 2023-11-17:
> Can you tell me how you solved the problem? The same error occurred to me.

The problem was supposed to be a memory issue, but it's been so long that I kind of forgot exactly how it was fixed. Maybe you can show me your code?

HVwang123, 2023-11-19:
> > Can you tell me how you solved the problem? The same error occurred to me.
> 
> The problem was supposed to be a memory issue, but it's been so long that I kind of forgot exactly how it was fixed. Maybe you can show me your code?

I solved this problem. Maybe it's because I compiled it wrong. Thank you.