ucb-bar/gemmini #275: Spike can't find 'gemmini' in 'libcustomext.so'

k0nze, 2023-01-25:
Hello,

I wanted to run `mvin_mvout-baremetal` as described in the README. When I run the command 

```
spike --extension=gemmini mvin_mvout-baremetal` 
```
from `$CHIPYARD_ROOT/generators/gemmini/software/gemmini-rocc-tests/build`  I get the following error message:
```
couldn't find extension 'gemmini' in shared library 'libcustomext.so'
```
I followed the installation instructions and I'm able to run `template` with the verilator simulator. What am I missing?

hngenc, 2023-01-25:
Can you make sure you ran `source env.sh` before running the Spike command?

Also, are you sure that the `./build-setup.sh esp-tools` command in the [installation instructions](https://github.com/ucb-bar/gemmini#installing-chipyard-and-spike) completed successfully?

k0nze, 2023-01-26:
Yes, I always run `source ./env.sh` before using chipyard (otherwise `./scripts/run-verilator.sh` wouldn't run). 
I reran `./build-setup.sh esp-tools` which completed successufully and now `spike` works. Thank you!

For reference:
The chipyard repository is at commit hash `004297b6a8c01be1b2110c4cf4f9393ae1ff8805` which corresponds to version `1.8.1`.
The gemmini repository is at commit hash `aaedc95247384dda63aa201d8d13492220df3cc3` which corresponds to version `v0.7.0`
`/toolchains/esp-tools/riscv-isa-sim/build` is at commit hash `051d820f08be84d069993de4375d29c91eb2f577` which corresponds to `$SPIKE_HASH`.

closed 2023-01-26 by k0nze