ucb-bar/gemmini #276: Counter documentation

Alexandre425, 2023-01-26:
Hi, I am developing benchmarking tools for gemmini, and I'm trying to make sense of which counters I should use in a number of cases.

Going by the counter names in the generated [ gemmini_counter.h](https://github.com/ucb-bar/gemmini-rocc-tests/blob/ae0cd8236d32fccf7197a7ac0634df5513cec4db/include/gemmini_counter.h), some are self explanatory, while others I have a hard time understanding what they're counting. 

Is there more detailed documentation on this, or at least some part of the source code that hints at it? I looked in the tutorial, but the information there regarding counters is very limited.

hngenc, 2023-01-28:
Sorry about that; we definitely need more documentation for the exact counters.

If you're OK with looking through the source code, then you can search for the counter names in the Scala files and find the exact signal that they are counting. For example, [this is where](https://github.com/ucb-bar/gemmini/blob/aaedc95247384dda63aa201d8d13492220df3cc3/src/main/scala/gemmini/ExecuteController.scala#L1012) the `A_GARBAGE_CYCLES` signal is wired up. But I'm not sure how useful that'll be.

We'll work on fully documenting the counters. In the meantime, are there specific counters in `gemmini_counter.h` that you're wondering about, or signals/events that you'd like to add counters for?

Alexandre425, 2023-01-29:
> We'll work on fully documenting the counters. 

Thank you, I appreciate it.

> In the meantime, are there specific counters in `gemmini_counter.h` that you're wondering about, or signals/events that you'd like to add counters for?

I'm trying to make sense of what the `MAIN_*_CYCLES` counters do, and how they differ from the `*_ACTIVE_CYCLE`. From my preliminary tests, it seems that the counters like `MAIN_LD_ST_CYCLES` etc., only increment when the two/three respective subsystems are operating simultaneously, is that correct?

hngenc, 2023-02-06:
> From my preliminary tests, it seems that the counters like MAIN_LD_ST_CYCLES etc., only increment when the two/three respective subsystems are operating simultaneously, is that correct?

Yes. We added those counters to try to measure how well we were overlapping instructions from different subsystems. Ideally, we want load, store, and execute instructions to be fully overlapped.

> I'm trying to make sense of what the MAIN_*_CYCLES counters do, and how they differ from the *_ACTIVE_CYCLE.

There is not a big difference; perhaps we should remove the `*_ACTIVE_CYCLE` counters to reduce complexity...

The only difference is that the `*_ACTIVE_CYCLES` are a little more accurate in terms of how much time the subsystems actually spend doing useful work. For example, the `LOAD_ACTIVE_CYCLES` won't increment if the LoadController is only updating config registers, but not actually issuing any load-requests to the outer memory system. But I imagine the actual difference in cycle count will not be very large.

Alexandre425, 2023-02-09:
Thank you, that's helpful.

While I don't have any more questions at this moment, they're bound to pop up as I keep exploring the accelerator. It would be very helpful if the existing counters were documented, even if it would be a comment over each line of [gemmini_counter.h](https://github.com/ucb-bar/gemmini-rocc-tests/blob/ae0cd8236d32fccf7197a7ac0634df5513cec4db/include/gemmini_counter.h) for a start.

hngenc, 2023-02-11:
Sure thing; we'll work on that. In the meantime, feel free to ask any other questions if they pop up later