ucb-bar/gemmini #286: [Feature Request] Support .tflite file and participate in the TinyML community

johnny-wang16, 2023-03-17:
Hi, as Gemmini gains more and more attention, I was wondering if you guys have plans to support the .tflite format. By support i mean enabling the functionality of deploying a .tflite format model to run on Gemmini. kind of like supporting an onnx format model to run on Gemmini.

fzi-peccia, 2023-08-29:
Hi! If you are interested in deploying TFLite to Gemmini, take a look at the integration I developed between TVM and Gemmini.

* https://discuss.tvm.apache.org/t/presenting-the-generation-of-code-for-the-gemmini-accelerator/14107

The code can be found here:

* https://github.com/fzi-peccia/tvm/tree/gemmini-microtvm

I also added a pull request here to add the Gemmini support for a non-zero padding, which is needed to deploy quantized TFLite models:

* https://github.com/ucb-bar/gemmini/pull/274

johnny-wang16, 2023-08-29:
Thanks for work! I'll take a look!