ucb-bar/gemmini #288: [error]

jasneetsinghwahan, 2023-03-22:
I am trying to practice lab 2 of EE290-2 HW for ML. 
I know that
> No SETUP or Debugging shall be supported for users outside the class
My request and proposal are that there are many like me trying to practice this lab. After completing this lab, I plan to make a video on setup and common pitfalls. So, many of us fellow outsides shall benefit and potentially not bug you. In the past, I have made detailed videos while practicing labs. [my videos link](https://www.youtube.com/@jasneetsinghwahan/about)

Machine details:
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.5 LTS
Release:	        20.04
Codename:	focal

After following these steps in this order:
`git clone https://github.com/ucb-bar/chipyard.git`
`cd chipyard`
`git checkout 1.8.1`
`./build-setup.sh esp-tools`
`source ./env.sh`
`cd generators/gemmini`
`git update submodule`
`git checkout ee290-sp21-lab2`

```
(/home/jasneets/chipyard/.conda-env) jasneets@jasneets-VirtualBox:~/chipyard/generators/gemmini/src/main/resources/vsrc$ git branch
* ee290-sp21-lab2
  master
```

```
(/home/jasneets/chipyard/.conda-env) jasneets@jasneets-VirtualBox:~/chipyard$ git branch
* (HEAD detached at 1.8.1)
  main

```
I just wanted to see if my environment and repos are properly setup, so instead of beginning with the code, I decided to elaborate on the design:

`cd ~/chipyard/sims/verilator/`
`make CONFIG=GemminiEE290Lab2RocketConfig`

But the errors I am receiving shouldn't come. A few of the errors are:
```
[error] /home/jasneets/chipyard/generators/gemmini/src/main/scala/gemmini/BeatMerger.scala:136:15: value toBool is not a member of chisel3.Reset
[error]   when (reset.toBool()) {
[error]               ^
```
I tried manually removing these errors by editing these 9 files and changing from toBool to asBool. But post that, I faced another set of 27 errors. **Please help**
[image]

hngenc, 2023-11-27:
Looking at your error message, it looks like you just have to keep replacing `toBool()` with `asBool`. Chisel and Chipyard were updated recently so that some older code became deprecated. It's annoying, but there's nothing to do but manually replace those function calls (or perhaps you could write a script to do so?)

jasneetsinghwahan, 2023-11-28:
> Looking at your error message, it looks like you just have to keep replacing `toBool()` with `asBool`. Chisel and Chipyard were updated recently so that some older code became deprecated. It's [...]

Thank You. it's helpful

closed 2023-11-28 by jasneetsinghwahan