ucb-bar/gemmini #291: About multi-Gemminis

kimjungwow, 2023-03-31:
Hello 😊

[image]

I would like to study an environment in which two Gemminis coexist, as in the IIWSC Tutorial p16.
How can I run two Gemminis together?

Also, is it possible for two Gemminis to divide and execute one application or can they only execute different applications?

Thank you!

hngenc, 2023-11-27:
You can set the number of CPUs that you [want here](https://github.com/ucb-bar/gemmini/blob/master/src/main/scala/gemmini/CustomSoCConfigs.scala#L21), and each CPU will have it's own Gemmini.

When running software on these Gemminis, you will want to make sure that you "pin" each process or thread to it's own CPU; because Gemmini can't "evict" the data in its scratchpads to another Gemmini when a thread moves to another CPU/Gemmini.

As for splitting models across multiple Gemminis, that's mainly a software issue. We have people working on building software for this on Onnx-Runtime, but support is currently pretty limited. For now, you might need to write your own software that splits a model (or a tensor) into two and assigns each half to a separate Gemmini.

Dignity-ghost, 2024-11-29:
Could you please provide some api examples to help users to write multiple-gemmini software? I mean the example of how to choose different gemminis. The exsisting apis in bareMetalC does not have the input of gemmini hardware ID.