ucb-bar/gemmini #295: About the waveforms generated by running ' ./scripts/run-verilator.sh  --debug'

JIAYouaman, 2023-05-12:
**Problem:**
I successfully generated a waveform(.vcd file) named 'waveform.vcd' using the command_line './scripts/build-verilator.sh --debug matmul' .But when i use GTKwave open it,I found some modules missing there.There is no name of the signal or the module related to 'gemmini'.Is it correct?If not,how to solve it?

Thanks!

**Other information:**
[image]

JIAYouaman, 2023-05-14:
It seems to be a meaningless issue. Sorry about that.I'm deeply aware that this is causing you trouble.
I have found the signals of Gemmini in  'TOP/TestHarness/chiptop/system/tile_prci_domain/tile_reset_domain/tile'

closed 2023-05-14 by JIAYouaman

hngenc, 2023-05-22:
No worries at all; glad you figured it out!

ymingX, 2024-06-03:
I want to know which version of chipyard and gemmini you are using. I checkout chipyard v1.11.0 and gemmini v0.7.2, an error occurred when running  './scripts/build-verilator.sh --debug' :

> VTestDriver__Trace__7__Slow.cpp: In function 'void VTestDriver___024root__trace_init_sub__TOP__14(VTestDriver___024root*, VerilatedVcd*)':VTestDriver__Trace__7__Slow.cpp:2678:131:
 error: expected unqualified-id before ',' token
 2678 |     tracep->declQuad(c+70308,0,"channel",-1, VerilatedTraceSigDirection::NONE, VerilatedTraceSigKind::VAR, VerilatedTraceSigType::, false,-1, 63,0);                                                                                                                                    
make[1]: *** [/mnt/e/Ubuntu2004/chipyard/.conda-env/share/verilator/include/verilated.mk:275：VTestDriver__Trace__7__Slow.o] 错误 1
make[1]: 离开目录“/mnt/e/Ubuntu2004/chipyard/sims/verilator/generated-src/chipyard.harness.TestHarness.CustomGemminiSoCConfig/chipyard.harness.TestHarness.CustomGemminiSoCConfig.debug”
make: *** [Makefile:212：/mnt/e/Ubuntu2004/chipyard/sims/verilator/simulator-chipyard.harness-CustomGemminiSoCConfig-debug] 错误 2

I tried to modified **VTestDriver__Trace__7__Slow.cpp:2678** to solve it. However, this file was auto-generated. When I rerun './scripts/build-verilator.sh --debug' , it was rewritten and my changes lost.