ucb-bar/gemmini #297: PE modification

babdary, 2023-05-20:
Hi, 
We have done some modifications on scala files about PEs for our purposes. However, we could not see any difference in the output while testing it with SPIKE. We ran build-spike.sh and setup-paths.sh to re-compile the modified .scala files. We wonder if we are using wrong script. Could you please help us about compiling our modified .scala file to the configuration?

hngenc, 2023-11-29:
Changing the `.scala` files won't have any impact on Spike, which is instead written in C++.

On the latest version of Gemmini, you can find the Spike implementation in `chipyard/generators/gemmini/software/libgemmini`. You'll probably want to update `gemmini.cc` to match your changes to the Scala/Chisel source files.

IoannouKon, 2024-03-27:
Hello 
, i think i have the same error i in the verilator run , i have changed PE but is seems like nothing happened 
 !the steps i follow  are 
1)change the scala src/main/ i  modified PE
2)./scripts/setup-paths.sh 
3)build gemmini-rocc-tests 
4) build spike 
5)build verilator 
6) run tests with spike and verilator  

But from results it seems like i never modified PE ,what can i do?