ucb-bar/gemmini #311: Filename/arguments mismatch between run-verilator.sh and actual verilator executable

jingkaih, 2023-08-28:
Hi, 

I ran into a issue that the executable doesn't exist when I ran 

```shell
./scirpts/run_verilator.sh --pk --debug template
``` 
which says that `./scripts/run-verilator.sh: line 91: ./simulator-chipyard-CustomGemminiSoCConfig: No such file or directory`
I looked at the _run_verilator.sh_ and it turned out that the executable generated was actually named as _simulator-chipyard.**harness**-CustomGemminiSoCConfig-debug_
So I modified the file name accordingly and made it work. 

However there comes the second issue. When debug mode is enabled, it appears _run_verilator.sh_  passes an `-v` followed by the [path](https://github.com/ucb-bar/gemmini/blob/master/scripts/run-verilator.sh#L68) of the waveform file to the executable.
However, an error message appeared: `./simulator-chipyard.harness-CustomGemminiSoCConfig-debug: invalid option -- 'v'`
I'm trying to manually run this executable to generate waveforms without scripting but I don't know the arguments it expects as the help message of this executable doesn't convey much. I think I'm able to run verilator without waveform using 
```shell
./simulator-chipyard.harness-CustomGemminiSoCConfig-debug pk -p /home/jingkai/chipyard/generators/gemmini/software/gemmini-rocc-tests/build/bareMetalC/template-pk
``` 
But I'm still struggling with figuring out the correct arguments for generating waveforms. Any help would be appreciated.

Chen-gz233, 2023-11-15:
hI,
I have the same issue on a Ubuntu 22.04. My logfile ends with

`(/home/chen/chipyard/.conda-env) chen@chen-WUJIE14-PRO:~/chipyard/generators/gemmini$ ./scripts/run-verilator.sh template
./scripts/run-verilator.sh: line 91: ./simulator-chipyard-CustomGemminiSoCConfig: No such file or directory
(/home/chen/chipyard/.conda-env) chen@chen-WUJIE14-PRO:~/chipyard/generators/gemmini$ cd ~
(/home/chen/chipyard/.conda-env) chen@chen-WUJIE14-PRO:~$ cd chipyard/
(/home/chen/chipyard/.conda-env) chen@chen-WUJIE14-PRO:~/chipyard$ source env.sh
(/home/chen/chipyard/.conda-env) chen@chen-WUJIE14-PRO:~/chipyard$ cd ..
(/home/chen/chipyard/.conda-env) chen@chen-WUJIE14-PRO:~$ cd chipyard/generators/gemmini
(/home/chen/chipyard/.conda-env) chen@chen-WUJIE14-PRO:~/chipyard/generators/gemmini$ ./scripts/run-verilator.sh template
./scripts/run-verilator.sh: line 91: ./simulator-chipyard-CustomGemminiSoCConfig: No such file or directory
(/home/chen/chipyard/.conda-env) chen@chen-WUJIE14-PRO:~/chipyard/generators/gemmini$ ./scirpts/run_verilator.sh --pk --debug template
-bash: ./scirpts/run_verilator.sh: No such file or directory
`

Did you solve this issue?

jingkaih, 2023-12-07:
> hI, I have the same issue on a Ubuntu 22.04. My logfile ends with
>
> `(/home/chen/chipyard/.conda-env) chen@chen-WUJIE14-PRO:~/chipyard/generators/gemmini$ ./scripts/run-verilator.sh template [...]

8u9mak3r, 2024-01-21:
> Hi, Please modify line 91 as I suggested. Something like simulator-chipyard.harness-CustomGemminiSoCConfig will help. As for dumping the waveforms using verilator, what works for me is going to chipyard/sims/verilator and running `make CONFIG=CustomGemminiSoCConfig run-binary-debug BINARY="home/jingkai/chipyard/generators/gemmini/software/gemmini-rocc-tests/build/bareMetalC/conv-baremetal"` However this is just a workaround not the solution though as it needs recompile the whole simulator every time you want to run some simple binary tests.
> 
> > hI, I have the same issue on a Ubuntu 22.04. My logfile ends with
> > `(/home/chen/chipyard/.conda-env) chen@chen-WUJIE14-PRO:~/chipyard/generators/gemmini$ ./scripts/run-verilator.sh template
> > ./scripts/run-verilator.sh: line 91: ./simulator-chipyard-CustomGemminiSoCConfig: No such file or directory
> > (/home/chen/chipyard/.conda-env) chen@chen-WUJIE14-PRO:~/chipyard/generators/gemmini$ cd ~
> > (/home/chen/chipyard/.conda-env) chen@chen-WUJIE14-PRO:~$ cd chipyard/
> > (/home/chen/chipyard/.conda-env) chen@chen-WUJIE14-PRO:~/chipyard$ source env.sh
> > (/home/chen/chipyard/.conda-env) chen@chen-WUJIE14-PRO:~/chipyard$ cd ..
> > (/home/chen/chipyard/.conda-env) chen@chen-WUJIE14-PRO:~$ cd chipyard/generators/gemmini
> > (/home/chen/chipyard/.conda-env) chen@chen-WUJIE14-PRO:~/chipyard/generators/gemmini$ ./scripts/run-verilator.sh template
> > ./scripts/run-verilator.sh: line 91: ./simulator-chipyard-CustomGemminiSoCConfig: No such file or directory
> > (/home/chen/chipyard/.conda-env) chen@chen-WUJIE14-PRO:~/chipyard/generators/gemmini$ ./scirpts/run_verilator.sh --pk --debug template
> > -bash: ./scirpts/run_verilator.sh: No such file or directory
> > `
> > Did you solve this issue?

Under this circumstance option 'debug' should follow '-' instead of '--', then it'll work fine.

8u9mak3r, 2024-01-21:
> Under this circumstance option 'debug' should follow '-' instead of '--', then it'll work fine.

And don't forget modify line 91. Replace 'chipyard' with 'chipyard.harness'

8u9mak3r, 2024-01-21:
> Under this circumstance option 'debug' should follow '-' instead of '--', then it'll work fine.

Well, it's not the right solution. run-verilator.sh ignores '-debug' without waveform generated. Why not asking @hngenc for help?