ucb-bar/gemmini #312: could not find chipyard.harness.TestHarness.CustomGemminiSoCConfig.top.v

Priyadharshiniganesh, 2023-09-10:
Hi,
The everything went well while building the Gemmini Hardware. Followed the tutorial and didn't find any errors, but I couldn't find chipyard.harness.TestHarness.CustomGemminiSoCConfig.top.v in the directory generated-src/chipyard.harness.TestHarness.CustomGemminiSoCConfig/gen-collateral. 

Here is the look at list of .v files inside this directory:
[image]

Please let me know , where I went wrong.

Regards,
Priya

hngenc, 2023-11-30:
Sorry, it looks like newer versions of Chipyard no longer generate the old `top.v` file. However, in `gen-collateral/`, you should be able to find `ChipTop.sv` and `Gemmni.sv`, which might be what you want.

BatCaveDev-PG, 2024-06-01:
Thank you this helped.