ucb-bar/gemmini #317: How can I see debugging information in hardware source files such as printf

ErgouGuan, 2023-11-28:
I want to mount the software at the hardware level and at the same time output the printf information of the hardware to monitor the related conditions of the hardware. What should I do? Thank you very much for your answers

 ./scripts/run-verilator.sh template

hngenc, 2023-11-29:
Did you put `printf`s in the Chisel source code, and you're trying to get them to print when you're running a simulation?

If so, try modifying [this line](https://github.com/ucb-bar/gemmini/blob/master/scripts/run-verilator.sh#L91) to:

```
./simulator-chipyard-CustomGemminiSoCConfig${DEBUG} +verbose $PK ${full_binary_path}
```

I believe the `+verbose` should cause `printf`s to work. Let me know if it doesn't work.

Just as a warning, this will also generate a _lot_ of output which you probably don't want, and which you will need to filter out.

ErgouGuan, 2023-11-29:
The explanation is very clear, it’s solved, thank you very much! !

hngenc, 2023-11-29:
No problem!