ucb-bar/gemmini #319: I didn‘t  change the hardware logic, but Test failed

ErgouGuan, 2023-12-20:
Hello, I would like to ask, on the premise that I did not change the hardware logic, I ran the following test output and reported an error. What may be the reason? Is it because I didn't turn on some parameter settings? Thank you very much for your answer

./scripts/run-verilator.sh conv_first_layer
*** FAILED *** (tohost = 1)
*** FAILED *** (code = 1, seed 1703053065) after 53768365 cycles
In addition, I would like to ask, if running this command fails, where will the log file be placed?
Thanks again！

mik300, 2024-05-04:
Hello, did you solve the issue? I'm running into the same problem.

1024GeBug, 2024-09-10:
> Hello, did you solve the issue? I'm running into the same problem.

Sorry I didn't notice your comment. After some research, I think it may be caused by the scratchpad being too small. Try to enlarge the scratchpad and run the test case again.

closed 2024-09-10 by 1024GeBug