ucb-bar/gemmini #326: error:Installing Chipyard and Spike

GGKOP, 2024-01-18:
I have been using Chipyard to generate the Sonic Boom portion without any issues. However, when attempting to install Gemmini using the following command line, I encountered problems.

```
git clone https://github.com/ucb-bar/chipyard.git
cd chipyard
git checkout 1.9.1
./build-setup.sh riscv-tools

source env.sh

cd generators/gemmini
git config remote.origin.fetch "+refs/heads/*:refs/remotes/origin/*"
git fetch && git checkout v0.7.1
git submodule update --init --recursive

make -C software/libgemmini install
```

```
(/home/ricky/chipyard/.conda-env) ricky@ricky-MS-7D70:~/chipyard/generators/gemmini$ 
make -C software/libgemmini install
make: Entering directory '/home/ricky/chipyard/generators/gemmini/software/libgemmini'
Running with RISCV=/home/ricky/chipyard/.conda-env/riscv-tools
g++ -L /home/ricky/chipyard/.conda-env/riscv-tools/lib -Wl,-rpath,/home/ricky/chipyard/.conda-env/riscv-tools/lib -shared -o libgemmini.so -std=c++17 -I /home/ricky/chipyard/.conda-env/riscv-tools/include -fPIC gemmini.cc -lriscv
In file included from gemmini.cc:1:
gemmini.h:4:10: fatal error: riscv/extension.h: No such file or directory
    4 | #include <riscv/extension.h>
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:12: libgemmini.so] Error 1
make: Leaving directory '/home/ricky/chipyard/generators/gemmini/software/libgemmini'
```

what can I do？

Zamion101, 2024-03-20:
I'm also getting same error while I'm following the official installation guide of the chipyard. This is most probably a new bug after some changes has been committed. 

```bash
==> Installing gemmini spike extensions
make: Entering directory '<redacted>/chipyard/generators/gemmini/software/libgemmini'
Running with RISCV=<redacted>/chipyard/.conda-env/riscv-tools
g++ -L /<redacted>/chipyard/.conda-env/riscv-tools/lib -Wl,-rpath,<redacted>/chipyard/.conda-env/riscv-tools/lib -shared -o libgemmini.so -std=c++17 -I <redacted>/chipyard/.conda-env/riscv-tools/include -fPIC gemmini.cc -lriscv
In file included from gemmini.cc:1:
gemmini.h:4:10: fatal error: riscv/extension.h: No such file or directory
 #include <riscv/extension.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:12: libgemmini.so] Error 1
```

Zamion101, 2024-04-22:
Do you have any update on this topic @GGKOP ?

jerryz123, 2024-04-22:
I'm not sure what instructions you are following for gemmini. After running chipyard's build-setup, no further steps are needed to run gemmini simulations

Zamion101, 2024-04-22:
I'm following instructions from [Initial-Repo-Setup](https://chipyard.readthedocs.io/en/stable/Chipyard-Basics/Initial-Repo-Setup.html#setting-up-the-chipyard-repo) which should work out of the box but I'm getting the above error when installing gemmini.

jerryz123, 2024-04-22:
I don't know what "installing gemmini" instructions you are referring to. But those are likely out of date or redundant. 

Any gemmini evaluations can be performed immediately after initial-repo-setup

Zamion101, 2024-04-22:
After clean clone of Chipyard, "Initial-Repo-Setup" tells to run `./build-setup.sh riscv-tools` to build Chipyard and all of the required dependencies which also contains `gemmini` and while "compiling" (sorry for the previous confusion) it gave an error. I tried multiple times to clear conda env and run again with clean chipyard. I have also cleaned conda completely and installed from stratch and tried with fresh & clean Chipyard but still the same problem. I have lost count of how many combinations have I tried so far.

jerryz123, 2024-04-22:
Do these instructions not work?

```
git clone https://github.com/ucb-bar/chipyard.git
cd chipyard
./build-setup.sh riscv-tools

source env.sh

make -C software/libgemmini install
```

Zamion101, 2024-04-22:
`./build-setup.sh riscv-tools` is the problem. It tries to compile `spike/gemmini` when executing `./scripts/build-toolchain-extra.sh`  https://github.com/ucb-bar/chipyard/blob/main/scripts/build-toolchain-extra.sh#L106-L112 and fails because of the above error.

GGKOP, 2024-04-22:
I'm sorry I didn't see your discussion in time. I had re-downloaded parts of Chipyard and Gemmini and then followed the commands from the official manual, which avoided the issue. In fact, I'm not sure what caused the problem now. It might have been related to the compiler.