ucb-bar/gemmini #328: error: gemmini verilator resnet50

GGKOP, 2024-01-23:
I used the default configuration of Gemmini and, after generating files in Verilator, I intended to execute resnet-pk on this simulator. However, the following error occurred. What should I do?

```
(/home/ricky/chipyard/.conda-env) ricky@ricky-MS-7D70:~/chipyard/sims/verilator$ ./simulator-chipyard-CustomGemminiSoCConfig pk resnet50-pk 3>&1 1>&2 2>&3>stats.txt
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 40027
```

stats.txt
```
bbl loader
[UART] UART0 is here (stdin/stdout).
[4133000] %Error: TLMonitor_48.sv:389: Assertion failed in TOP.TestHarness.chiptop.system.tile_prci_domain.tile_reset_domain_tile.gemmini.spad.buffer.monitor: Assertion failed: 'A' channel carries Get type which slave claims it can't support (connected at Scratchpad.scala:200:27)
    at Monitor.scala:42 assert(cond, message)

%Error: /home/ricky/chipyard/sims/verilator/generated-src/chipyard.TestHarness.CustomGemminiSoCConfig/gen-collateral/TLMonitor_48.sv:389: Verilog $stop
Aborting...
```

GGKOP, 2024-01-24:
@hngenc @jerryz123  Could you please assist me? I would greatly appreciate your help.

Chen-gz233, 2024-04-25:
hi,did you solve this problem?

GGKOP, 2024-04-25:
No，I give up

mik300, 2024-12-16:
Any luck resolving the issue?