ucb-bar/gemmini #341: The float32 configuration fails to pass even the simplest test in Verilator

chguo1995, 2024-03-27:
I attempted to use the 8*8 float32 configuration as follows:

```
val trainingConfig = defaultFpConfig.copy(
  inputType = Float(expWidth = 8, sigWidth = 24),
  accType = Float(expWidth = 8, sigWidth = 24),

  meshRows = 8,
  meshColumns = 8,
  
  dataflow = Dataflow.WS,

  has_training_convs = true,
  has_max_pool =  false,

  sp_capacity = CapacityInKilobytes(512),
  acc_capacity = CapacityInKilobytes(256),
)
```

However, it fails to pass the _Verilator_ simulation, even for a simple **mvin_mvout.c** test, which gets stuck.

Additionally, in **configsFP.scala**, _spad_read_delay_ was set to **1**. I attempted to change _spad_read_delay_ to **4** (to align with the int8 configuration), which allowed it to pass the **mvin_mvout.c** test and also all the _DIM*DIM_ computations. However, when executing computations related to **tile_matmul.c** or **tile_conv.c**(i.e., involving a series of instructions related to _loop_matmul_ws_ and _loop_conv_ws_), it fails to produce correct results. 

Tested versions include: **v0.7.0, v0.7.1, v0.7.2, and the development version.**

closed 2024-03-31 by chguo1995

1024GeBug, 2024-12-03:
Hello, I would like to ask a question. Does Gemmini support the classic FP32 format, which consists of 1 sign bit, 8 exponent bits, and 23 mantissa bits, or does it only support 8 exponent bits and 24 mantissa bits? I have encountered a similar issue when building Gemmini. I would greatly appreciate any help you can provide.

1024GeBug, 2024-12-03:
Thank you very much !!!!!!!

hansa31, 2025-01-16:
Hi! How did you manage to get correct results for tile_matmul.c?