ucb-bar/gemmini #342: Changing the value of scratchpad does not change the cycles run by workloads. Why is this?

ArchRookie, 2024-03-28:
I changed the scratchpad value many times from 64 to 128 to 256, but the cycles run by the simulated workloads will not change. Why is this? Can you solve my doubts? I will be very grateful.

hngenc, 2024-03-28:
In `gemmini.h`, tiling factors will be calculated for your workloads. For example, `tiled_matmul_auto` calculates tiling factors for matrix multiplications. You should check what your tiling factors are with different sizes for your scratchpad and accumulator. If they are the same, then your performance won't change, because `tiled_matmul_auto` will utilize the same scratchpad capacity even with different scratchpad sizes.

ArchRookie, 2024-03-28:
Can you explain it in more detail? Really grateful!

ArchRookie, 2024-03-29:
> In `gemmini.h`, tiling factors will be calculated for your workloads. For example, `tiled_matmul_auto` calculates tiling factors for matrix multiplications. You should check what your tiling factors [...]

Can you explain it in more detail? Really grateful!

hngenc, 2024-03-29:
> Can you explain it in more detail?

Basically, you want to print out [these tiling factors.](https://github.com/ucb-bar/gemmini-rocc-tests/blob/dev/include/gemmini.h#L1310-L1327)

They determine how many rows of your scratchpad/accumulator SRAMs will be used in each tile of the tiled matrix multiplications.

ArchRookie, 2024-03-29:
Thank you very much!

> Basically, you want to print out [these tiling factors.](https://github.com/ucb-bar/gemmini-rocc-tests/blob/dev/include/gemmini.h#L1310-L1327)
>
> They determine how many rows of your [...]

closed 2024-04-25 by ArchRookie