ucb-bar/gemmini #343: CHANGE HARDWARE

IoannouKon, 2024-03-29:
Hello, 
I have modified the MAC in PE so i can use approximate design to save area and energy.I want to check if my Modified work in gemmini .i have build the verilator and spike again after that.But in spike  i think nothing change and the verilator takes too much time to run .Why the spike cant "see:" the changes i have done in scala?

1024GeBug, 2024-12-03:
The issue is that Spike is written in C and C++, and it’s a software simulation rather than hardware. Therefore, when you try to run your written and compiled hardware, along with custom instructions, on Spike, there’s no response.

I am facing a similar challenge. I have added some features to Gemmini and want to test them on real datasets. However, it seems that ResNet-50 (the provided example) can only run on Spike. Do you have any solutions for this?

IoannouKon, 2024-12-03:
Hello , 

**Short Answer:** It is probably not possible.

**Explanation:**  
Spike does support running C simulations for some hardware, such as Gemmini. However, I don't think that if you manually change the hardware, the C simulation will automatically adapt to reflect those changes. This means that your modifications will never be fully simulated with Spike. Personally, I use Verilator for these cases.

What I did was use Verilator to simulate the custom hardware accelerator I built. While Verilator is certainly much slower (depending on the computational system you run it on), it provides good accuracy, especially if you care about the execution cycles of the accelerator.

If you find a solution for simulating custom hardware using Spike (in C), please let me know!

1024GeBug, 2024-12-04:
> Hello ,
>
> **Short Answer:** It is probably not possible.
>
> **Explanation:** Spike does support running C simulations for some hardware, such as Gemmini. However, I don't think that if you manually [...]

Additionally, I would like to ask if it's possible to modify the Spike workloads in a way that allows them to run on Verilator? Do you have any experience with this?