ucb-bar/gemmini #358: Error while building verilator with waveforms.

ymingX, 2024-06-03:
I git clone and checkout chipyard v1.11.0 and gemmini v0.7.2, successfully configuring chipyard by running "./build-setup.sh riscv-tools" as chipyard docs. Then I tried to set up gemmini as readme.md. 
I successfully built and run verilator. Further, when following readme running  "if you want a simulator that can generate waveforms, run this:
` ./scripts/build-verilator.sh --debug` 

An error occurred:

> VTestDriver__Trace__7__Slow.cpp: In function 'void VTestDriver___024root__trace_init_sub__TOP__14(VTestDriver___024root*, VerilatedVcd*)':
VTestDriver__Trace__7__Slow.cpp:2678:131: error: expected unqualified-id before ',' token
 2678 |     tracep->declQuad(c+70308,0,"channel",-1, VerilatedTraceSigDirection::NONE, VerilatedTraceSigKind::VAR, VerilatedTraceSigType::, false,-1, 63,0);
      |                                                                                                                                   ^
make[1]: *** [/mnt/e/Ubuntu2004/chipyard/.conda-env/share/verilator/include/verilated.mk:275：VTestDriver__Trace__7__Slow.o] 错误 1
make[1]: 离开目录“/mnt/e/Ubuntu2004/chipyard/sims/verilator/generated-src/chipyard.harness.TestHarness.CustomGemminiSoCConfig/chipyard.harness.TestHarness.CustomGemminiSoCConfig.debug”
make: *** [Makefile:212：/mnt/e/Ubuntu2004/chipyard/sims/verilator/simulator-chipyard.harness-CustomGemminiSoCConfig-debug] 错误 2

I think there may be a missing object so I modified it to this like other lines of this file:

 tracep->declQuad(c+70308,0,"channel",-1, VerilatedTraceSigDirection::NONE, VerilatedTraceSigKind::VAR, VerilatedTraceSigType::**LOGIC**, false,-1, 63,0);

However, this file was auto-generated, my changes lost when running ` ./scripts/build-verilator.sh --debug`  again so that this error still occurred.

detailed log is attached here:
[build-debugbak.log](https://github.com/user-attachments/files/15528270/build-debugbak.log)

AnjelicaB, 2024-06-26:
I have the same error, did you figure it out?

ymingX, 2024-07-06:
> I have the same error, did you figure it out?

no