ucb-bar/gemmini #359: Using a newer chipyard/firesim version?

AnjelicaB, 2024-06-12:
I was just wondering if there's a particular reason that the readme said to use the Chipyard 1.9.1 version, instead of newer versions (eg 1.10 or 1.11). Additionally, if I want to run FireSim with Gemmini on a U280 board, I need the version of Firesim that is 1.17 +.

Could I generate a Gemmini accelerator using Chipyard 1.11 and run a simulation on Firesim 1.18 with Xilinx U280 board?

jerryz123, 2024-06-13:
The README in this repo has not been updated in a while. 

Generally, the versions of all the projects referenced by chipyard/main will work together, and will support recent features in chipyard/firesim.

AnjelicaB, 2024-06-25:
Gemmini build verilator fails if I clone into Chipyard 1.11.0 (instead of 1.9.1), then follow the steps:

`cd generators/gemmini
git config remote.origin.fetch "+refs/heads/*:refs/remotes/origin/*"
git fetch && git checkout v0.7.1
git submodule update --init --recursive `

It fails with version conflicts.

Should I skip this step completely?

shirohasuki, 2024-09-09:
@AnjelicaB I'm having the same problem, I think it's due to chipyard version upgrade (previously it was fine to run verilator scripts in chipyard 1.9.1). Did you solve the problem?

AnjelicaB, 2024-09-09:
No unfortunately. The best I could do is to use Chipyard 1.11, and leave
everything in there as is. I could build Gemmini software, I just couldn’t
run it on Spike or Verilator