ucb-bar/gemmini #367: build-setup.sh riscv-tools taking too long to compile

svs11, 2024-10-27:
Hi, I am trying to use Gemmini and set up the environment for the accelerator. The setup is stuck at the step "./build-setup.sh riscv-tools". It has been running in the background for more than 48 hours, executing "ctags -R --exclude=@.ctagsignore --links=no". Is it normal for this script to take that long?

svs11, 2024-10-28:
got it fixed had issue with the system.

closed 2024-10-28 by svs11

linbaiwpi, 2024-11-11:
may I have the Chipyard and Gemmini as well as Verilator and Chisel version of your configuration? I cannot build Gemmini sucessfully.

> got it fixed had issue with the system.

svs11, 2024-11-12:
hi , my chipyard version:  1.9.1, gemmini :  v0.7.1, Verilator : 4.226 and SBT Versions: sbt version in this project: 1.3.13
sbt script version: 1.8.2. If you are facing the same problem then remove the tags directory and re-run , if fails again then try to remove the whole directory and build it again.