ucb-bar/gemmini #368: Error when running `./scripts/build-verilator.sh`

jona1115, 2024-11-01:
I am following the tutorial on the README.md to set up gemmini and I encounter this error when running the `./scripts/build-verilator.sh` script:

```
(/home/vm-user/chipyard/.conda-env) vm-user@cpre581-duwe-base:~/chipyard/generators/gemmini ((HEAD detached at v0.7.2))$ ./scripts/build-verilator.sh
Running with RISCV=/home/vm-user/chipyard/.conda-env/riscv-tools
echo "Checking all submodules in generators/ are initialized. Uninitialized submodules will be displayed" ; ! git submodule status /home/vm-user/chipyard/generators | grep ^-
Checking all submodules in generators/ are initialized. Uninitialized submodules will be displayed
mkdir -p /home/vm-user/chipyard/.classpath_cache/
cd /home/vm-user/chipyard && java -jar /home/vm-user/chipyard/scripts/sbt-launch.jar -Dsbt.ivy.home=/home/vm-user/chipyard/.ivy2 -Dsbt.global.base=/home/vm-user/chipyard/.sbt -Dsbt.boot.directory=/home/vm-user/chipyard/.sbt/boot/ -Dsbt.color=always -Dsbt.supershell=false -Dsbt.server.forcestart=true ";project chipyard; set assembly / assemblyOutputPath := file(\"/home/vm-user/chipyard/.classpath_cache/chipyard.jar\"); assembly" && touch /home/vm-user/chipyard/.classpath_cache/chipyard.jar
Picked up JAVA_TOOL_OPTIONS: -Xmx8G -Xss8M -Djava.io.tmpdir=/home/vm-user/chipyard/.java_tmp
[info] welcome to sbt 1.8.2 (N/A Java 20.0.2-internal)
[info] loading settings for project chipyard-build from plugins.sbt ...
[info] loading project definition from /home/vm-user/chipyard/project
[info] loading settings for project chipyardRoot from build.sbt ...
[info] loading settings for project testchipip from build.sbt ...
[info] loading settings for project barf from build.sbt ...
[info] loading settings for project constellation from build.sbt ...
[info] loading settings for project icenet from build.sbt ...
[info] loading settings for project cva6 from build.sbt ...
[info] loading settings for project ibex from build.sbt ...
[info] loading settings for project sodor from build.sbt ...
[info] loading settings for project midas_target_utils from build.sbt ...
[info] resolving key references (46535 settings) ...
[info] set current project to chipyardRoot (in build file:/home/vm-user/chipyard/)
[info] set current project to chipyard (in build file:/home/vm-user/chipyard/)
[info] Defining assembly / assemblyOutputPath
[info] The new value will be used by assembly
[info] Reapplying settings...
[info] set current project to chipyard (in build file:/home/vm-user/chipyard/)
Warning: Tried to extract the base path for relative glob tools/stage/src/main/scala/**/*.{java,scala}. To disable this warning, re-run the program with java option, -Dsbt.io.implicit.relative.glob.conversion=allow
Warning: Tried to extract the base path for relative glob tools/stage/src/main/scala/**/*.{java,scala}. To disable this warning, re-run the program with java option, -Dsbt.io.implicit.relative.glob.conversion=allow
Warning: Tried to extract the base path for relative glob tools/firrtl2/src/main/java/**/*.{java,scala}. To disable this warning, re-run the program with java option, -Dsbt.io.implicit.relative.glob.conversion=allow
Warning: Tried to extract the base path for relative glob tools/firrtl2/src/main/java/**/*.{java,scala}. To disable this warning, re-run the program with java option, -Dsbt.io.implicit.relative.glob.conversion=allow
[info] compiling 13 Scala sources to /home/vm-user/chipyard/generators/gemmini/src/target/scala-2.13/classes ...
[error] /home/vm-user/chipyard/generators/gemmini/src/main/scala/gemmini/Normalizer.scala:5:8: object ChiselEnum is not a member of package chisel3.experimental
[error] import chisel3.experimental.ChiselEnum
[error]        ^
[error] /home/vm-user/chipyard/generators/gemmini/src/main/scala/gemmini/Configs.scala:7:8: object XLen is not a member of package freechips.rocketchip.tile
[error] import freechips.rocketchip.tile.{BuildRoCC, OpcodeSet, XLen}
[error]        ^
[error] /home/vm-user/chipyard/generators/gemmini/src/main/scala/gemmini/DMA.scala:6:8: object DataMirror is not a member of package chisel3.experimental
[error] import chisel3.experimental.DataMirror
[error]        ^
[error] /home/vm-user/chipyard/generators/gemmini/src/main/scala/gemmini/NormCmd.scala:6:8: object ChiselEnum is not a member of package chisel3.experimental
[error] import chisel3.experimental.ChiselEnum
[error]        ^
[error] /home/vm-user/chipyard/generators/gemmini/src/main/scala/gemmini/Controller.scala:35:16: not found: value XLen
[error]   val xLen = p(XLen)
[error]                ^
[error] /home/vm-user/chipyard/generators/gemmini/src/main/scala/gemmini/Controller.scala:406:16: overloaded method apply with alternatives:
...
[warn] one warning found
[error] 41 errors found
[error] (gemmini / Compile / compileIncremental) Compilation failed
[error] Total time: 16 s, completed Nov 1, 2024, 2:45:15 AM
make: *** [/home/vm-user/chipyard/common.mk:118: /home/vm-user/chipyard/.classpath_cache/chipyard.jar] Error 1
```

Does anyone have any idea?

suhwanhong999, 2024-12-02:
I am having very similar issues with Chipyard 1.9.1 and Gemmini 0.7.2
[image]

jona1115, 2024-12-02:
This issue is solved here [https://groups.google.com/g/chipyard/c/IKBsDogZzUQ](https://groups.google.com/g/chipyard/c/IKBsDogZzUQ)

suhwanhong999, 2024-12-03:
Thank you so much!

suhwanhong999, 2024-12-09:
For anyone looking, check out Commit #370 and NOT #369
https://github.com/ucb-bar/gemmini/pull/370/commits/46158fbdaaa5ff9ee89ce296ed4e88262338f599

yoyo115956, 2025-03-07:
Thank u, guys! that really help a lot.

Chipyard version need to match the Gemmini version.
https://github.com/ucb-bar/gemmini/tree/v0.7.1 
My chipyard is 1.9.1, so I check history version and find the corresponding version of gemmini v0.7.1, which would match and the README.md inside make it very clear.