ucb-bar/gemmini #374: Running Modified Gemmini Hardware with Custom Instructions on Verilator: Issues for ResNet-50 Benchmark

1024GeBug, 2024-12-03:
We have added some functional modules and a series of custom instructions to Gemmini. Now, we want to test the performance of these modules under real workloads (such as ResNet-50, etc.), but the provided benchmark tests for real models are based on Spike, which are not useful after modifying the hardware. However, when we try to run these cases directly using Verilator, we encounter various errors (i.e., fails).

So, my question is: how can we run these cases (e.g., ResNet-50) with Verilator? Any guidance or suggestions would be greatly appreciated!

1024GeBug, 2024-12-04:
Thank you for your attention and response.

What I mean to express is:

In Gemmini, the original benchmark provided some real model tests. However, these models can only run on Spike, with the original command being: ./scripts/run-spike.sh resnet50. I have added some functional modules to the hardware, and using Spike for simulation will not reflect these changes.

So what I need to do now is figure out how to run these simulations, originally intended for Spike, on Verilator. However, when I try to run them on Verilator, the simulation runs for a while and then hangs at some point, eventually failing.

How should I adjust things to be able to run real models on Verilator? Or are there other existing workloads I can use?

> Have you tried running the unmodified Gemmini design with Verilator to ensure the testbench and setup work correctly? If the unmodified design runs fine but the modified one fails, it’s likely an issue with your modifications.
> 
> What kind of error appears? Does it fail immediately or during execution? If it’s during execution, double-check your design changes for incorrect signal connections or timing issues. Sharing the error logs could help identify the problem.

Thank you for your attention and response.

What I mean to express is:

In Gemmini, the original benchmark provided some real model tests. However, these models can only run on Spike, with the original command being: ./scripts/run-spike.sh resnet50. I have added some functional modules to the hardware, and using Spike for simulation will not reflect these changes.

So what I need to do now is figure out how to run these simulations, originally intended for Spike, on Verilator. However, when I try to run them on Verilator, the simulation runs for a while and then hangs at some point, eventually failing.

How should I adjust things to be able to run real models on Verilator? Or are there other existing workloads I can use?

1024GeBug, 2024-12-04:
> Have you tried running the unmodified Gemmini design with Verilator to ensure the testbench and setup work correctly? If the unmodified design runs fine but the modified one fails, it’s likely an [...]

Thank you for your attention and response.

What I mean to express is:

In Gemmini, the original benchmark provided some real model tests. However, these models can only run on Spike, with the original command being: ./scripts/run-spike.sh resnet50. I have added some functional modules to the hardware, and using Spike for simulation will not reflect these changes.

So what I need to do now is figure out how to run these simulations, originally intended for Spike, on Verilator. However, when I try to run them on Verilator, the simulation runs for a while and then hangs at some point, eventually failing.

How should I adjust things to be able to run real models on Verilator? Or are there other existing workloads I can use?

IoannouKon, 2024-12-04:
> So what I need to do now is figure out how to run these simulations, originally intended for Spike, on Verilator. However, when I try to run them on Verilator, the simulation runs for a while and [...]
 
Hello again ! 
Are you running the test bench with default design or with a modification ? If ventilator fails with default design , then you probably need  a new test bebench ,if the configuration of the design is  correct.  Otherwise you modification is the problem and you need to find what is wrong .

1024GeBug, 2024-12-04:
Thank you for your response again!

I am using the ResNet-50 provided by Gemmini, and the command given on the official website is to use Spike for validation. I want to run it on Verilator, but I haven't made any changes to the workload. I tried running the original version of Gemmini (i.e., without any modifications to the hardware) on ResNet-50 using Verilator, but it actually results in an error.

In other words, I haven’t made any changes to the workload itself; the only change I made was switching from Spike simulation to Verilator simulation, and that's where the error occurs. I’m now trying to resolve this issue.

> Hello again ! Are you running the test bench with default design or with a modification ? If ventilator fails with default design , then you probably need a new test bebench ,if the configuration of [...]

1024GeBug, 2024-12-04:
> Hello again ! Are you running the test bench with default design or with a modification ? If ventilator fails with default design , then you probably need a new test bebench ,if the configuration of [...]

emmm, I'm not sure if I explained it clearly.

To put it simply, I want to take the workload that works fine when run with Spike and try to run it on Verilator, but it keeps throwing errors. I haven't been able to find a solution for this. I haven't modified the Gemmini hardware or the workload itself.

suhwanhong999, 2024-12-09:
Have you tried using the latest instructions? 
I've tried to get Gemmini to work for a month and after following this I finally got everything to work correctly. 
https://github.com/ucb-bar/gemmini/pull/370/commits/46158fbdaaa5ff9ee89ce296ed4e88262338f599

1024GeBug, 2024-12-10:
> Have you tried using the latest instructions? I've tried to get Gemmini to work for a month and after following this I finally got everything to work correctly. [46158fb](https://github.com/ucb-bar/gemmini/commit/46158fbdaaa5ff9ee89ce296ed4e88262338f599)

Thank you, the issue has been resolved. However, currently, the cycle count in Verilator is much higher than that in Spike.

closed 2024-12-10 by 1024GeBug