ucb-bar/gemmini #376: facing error at this stage "./scripts/build-verilator.sh"

sking-007, 2024-12-09:
error] /home/umair/Documents/STUDY/paperwerk/ge/chipyard/generators/chipyard/src/main/scala/config/GemminiCPUConfigs.scala:3:13: object common is not a member of package boom
[error] import boom.common._
[error]             ^
[error] /home/umair/Documents/STUDY/paperwerk/ge/chipyard/generators/chipyard/src/main/scala/config/GemminiCPUConfigs.scala:8:25: not found: type WithNBigCores
[error]   type RocketBigCores = WithNBigCores
[error]                         ^
[error] /home/umair/Documents/STUDY/paperwerk/ge/chipyard/generators/chipyard/src/main/scala/config/GemminiCPUConfigs.scala:9:25: not found: type WithNMedCores
[error]   type RocketMedCores = WithNMedCores
[error]                         ^
[error] /home/umair/Documents/STUDY/paperwerk/ge/chipyard/generators/chipyard/src/main/scala/config/GemminiCPUConfigs.scala:10:27: not found: type WithNSmallCores
[error]   type RocketSmallCores = WithNSmallCores
[error]                           ^
[error] /home/umair/Documents/STUDY/paperwerk/ge/chipyard/generators/chipyard/src/main/scala/config/GemminiCPUConfigs.scala:12:25: not found: type WithNLargeBooms
[error]   type BoomLargeCores = WithNLargeBooms
[error]                         ^
[error] /home/umair/Documents/STUDY/paperwerk/ge/chipyard/generators/chipyard/src/main/scala/config/GemminiCPUConfigs.scala:13:23: not found: type WithNMediumBooms
[error]   type BoomMedCores = WithNMediumBooms
[error]                       ^
[error] /home/umair/Documents/STUDY/paperwerk/ge/chipyard/generators/chipyard/src/main/scala/config/GemminiCPUConfigs.scala:14:25: not found: type WithNMediumBooms
[error]   type BoomSmallCores = WithNMediumBooms
[error]                         ^
[error] 7 errors found
[error] (Compile / compileIncremental) Compilation failed
[error] Total time: 35 s, completed Dec 9, 2024, 12:43:21 PM

suhwanhong999, 2024-12-09:
I've found that setup can be quite finnicky sometimes, I would suggest that you restart fresh. Also, don't run 'conda activate base', just download the lastest chipyard repo and run the setup script. At least that's how I resolved this error.

jerryz123, 2024-12-09:
The scripts in the scripts/ directory of gemmini have been unmaintained. However, gemmini's integration into Chipyard js regularly tested.

sking-007, 2024-12-10:
> I've found that setup can be quite finnicky sometimes, I would suggest that you restart fresh. Also, don't run 'conda activate base', just download the lastest chipyard repo and run the setup script. At least that's how I resolved this error.
> 
> Also, using ./build-verilator.sh is outdated as far as I am aware, check out the instructions in this README from the Gemmini repo [46158fb](https://github.com/ucb-bar/gemmini/commit/46158fbdaaa5ff9ee89ce296ed4e88262338f599) Edit: Clarity

i followed your instruction but error is still there.

sking-007, 2024-12-11:
yes im using ec2 instance ubuntu image t2.large nvida gpu

suhwanhong999, 2024-12-11:
If this is a fresh restart of the EC2 instance (make sure it is), can you list the commands you ran step by step?

sking-007, 2024-12-11:
fresh ec2 and commands
sudo apt update
apt upgrade
curl -L -O "https://github.com/conda-forge/miniforge/releases/latest/download/Miniforge3-$(uname)-$(uname -m).sh"
bash Miniforge3-$(uname)-$(uname -m).sh
source ~/.bashrc  # Or source ~/.zshrc for zsh users
conda --version
conda update -n base -c defaults conda
git clone https://github.com/ucb-bar/chipyard.git
cd chipyard
./build-setup.sh   
source env.sh
cd chipyard/sims/verilator
make CONFIG=YourGemminiConfig verilog

suhwanhong999, 2024-12-11:
You should follow the instructions in the link i sent earlier verbatim. 
More specifically, after you launc the instance, all you need to run is:
git clone https://github.com/ucb-bar/chipyard.git
cd chipyard
./build-setup.sh
source env.sh
cd chipyard/sims/verilator
make

sking-007, 2024-12-11:
./build-setup.sh by this command at last stage

BEGINNING STEP 9: Pre-compiling FireMarshal buildroot sources ==========
To check on progress, either call marshal with '-v' or see the live output at: 
/home/ubuntu/chipyard/software/firemarshal/logs/br-base-build-2024-12-11--15-18-43-UZUOP16BKQUVRGDE.log
.  /home/ubuntu/chipyard/software/firemarshal/boards/firechip/base-workloads/br-base/host-init.sh 
.  /home/ubuntu/chipyard/software/firemarshal/images/firechip/br.7251/br.7251.img
Attempting to download cached image: https://raw.githubusercontent.com/firesim/firemarshal-public-br-images/main/images/firechip/br.7251/br.7251.img.zip
TaskError - taskid:/home/ubuntu/chipyard/software/firemarshal/images/firechip/br.7251/br.7251.img
PythonAction Error
Traceback (most recent call last):
  File "/home/ubuntu/chipyard/.conda-env/lib/python3.10/site-packages/doit/action.py", line 461, in execute
    returned_value = self.py_callable(*self.args, **kwargs)
  File "/home/ubuntu/chipyard/software/firemarshal/boards/firechip/distros/br/br.py", line 271, in buildBaseImage
    wlutil.run(['make'], cwd=br_dir / "buildroot", env=env)
  File "/home/ubuntu/chipyard/software/firemarshal/wlutil/wlutil.py", line 527, in run
    raise sp.CalledProcessError(p.returncode, prettyCmd)
subprocess.CalledProcessError: Command 'make' returned non-zero exit status 2.

ERROR: Failed to build workload br-base.json
Log available at: /home/ubuntu/chipyard/software/firemarshal/logs/br-base-build-2024-12-11--15-18-43-UZUOP16BKQUVRGDE.log
ERROR: FAILURE: 1 builds failed
build-setup.sh: Build script failed with exit code 1 at step 9: Pre-compiling FireMarshal buildroot sources

sking-007, 2024-12-11:
final error

rogram with java option, -Dsbt.io.implicit.relative.glob.conversion=allow
[info] compiling 2 Scala sources to /home/ubuntu/chipyard/generators/chipyard/target/scala-2.13/classes ...
[error] /home/ubuntu/chipyard/generators/chipyard/src/main/scala/config/GemminiCPUConfigs.scala:3:13: object common is not a member of package boom
[error] import boom.common._
[error]             ^
[error] /home/ubuntu/chipyard/generators/chipyard/src/main/scala/config/GemminiCPUConfigs.scala:8:25: not found: type WithNBigCores
[error]   type RocketBigCores = WithNBigCores
[error]                         ^
[error] /home/ubuntu/chipyard/generators/chipyard/src/main/scala/config/GemminiCPUConfigs.scala:9:25: not found: type WithNMedCores
[error]   type RocketMedCores = WithNMedCores
[error]                         ^
[error] /home/ubuntu/chipyard/generators/chipyard/src/main/scala/config/GemminiCPUConfigs.scala:10:27: not found: type WithNSmallCores
[error]   type RocketSmallCores = WithNSmallCores
[error]                           ^
[error] /home/ubuntu/chipyard/generators/chipyard/src/main/scala/config/GemminiCPUConfigs.scala:12:25: not found: type WithNLargeBooms
[error]   type BoomLargeCores = WithNLargeBooms
[error]                         ^
[error] /home/ubuntu/chipyard/generators/chipyard/src/main/scala/config/GemminiCPUConfigs.scala:13:23: not found: type WithNMediumBooms
[error]   type BoomMedCores = WithNMediumBooms
[error]                       ^
[error] /home/ubuntu/chipyard/generators/chipyard/src/main/scala/config/GemminiCPUConfigs.scala:14:25: not found: type WithNMediumBooms
[error]   type BoomSmallCores = WithNMediumBooms
[error]                         ^
[error] 7 errors found
[error] (Compile / compileIncremental) Compilation failed
[error] Total time: 21 s, completed Dec 11, 2024, 6:44:14 PM

jerryz123, 2024-12-13:
The scripts in that directory are no longer supported

closed 2024-12-13 by jerryz123