ucb-bar/gemmini #393: Porting Gemmini RoCC to another CPU (requirements / behaviour)

Dolu1990, 2025-09-09:
Hi,

We are porting Gemmini to a CPU (not rocket / boom).
We got it to work well in baremetal <3

Thing is, i didn't found much information about the expected behaviour of the RoCC busy / interrupt flags / linux integration / ptw exception.

So far, my understanding is :
- CPU fence instruction should wait on the busy flag being low, to ensure memory coherency.

But i'm not sure how RoCC/Gemmini handle ptw exception / page fault, when running in linux.
- Did i missed some documentation ?
- Is there some special linux drivers or changes related to RoCC / gemmini   
- Should busy=1 prevent CPU privilege changes / SATP changes ?

Thanks ^^
Charles

Dolu1990, 2025-09-10:
Digging in the gemmini tests, i can see in most tests there is : 

```c
mlockall(MCL_CURRENT | MCL_FUTURE)
```

So, it seems that gemmini expect to never hit an unpopulated page, so it ask the kernel to lock them all (mlockall without MCL_ONFAULT)

Is it right ?

jerryz123, 2025-09-10:
That is correct